STMicroelectronics/STM32CubeWB #1: Example PKA: HAL_PKA_ECCMulFastMode

dmazzella, 2019-11-15:
Is it possible to have an example of PKA_ECCscalarMultiplication that uses HAL_PKA_ECCMulFastMode instead of HAL_PKA_ECCMul?

Thanks

ALABSTM, 2019-11-21:
Hi Damiano,

Your request will be transmitted to our development teams to see whether they could address it. We will be back to you as soon as we get an answer. Thank you in advance for your patience.

With regards,

dmazzella, 2019-11-21:
Hi Ali,
Thanks, I waiting news.

Regards.

PierreLeCorre, 2019-11-25:
Hello @dmazzella 

The behavior of `HAL_PKA_ECCMul` is described inside Projects/P-NUCLEO-WB55.Nucleo/Examples/PKA/PKA_ECCscalarMultiplication/Src/main.c.
The implementation done inside STM32Cube_FW_WB_V1.3.0 is as follow and demonstrate the `HAL_PKA_ECCMul` api usage.
```
/* USER CODE BEGIN PV */
PKA_HandleTypeDef hpka;
PKA_ECCMulInTypeDef in = {0};
PKA_ECCMulOutTypeDef out = {0};
__IO uint32_t operationComplete = 0;
/* USER CODE END PV */
```

```
  /* USER CODE BEGIN 2 */
  /* Set input parameters */
  in.modulusSize =     prime256v1_Prime_len;
  in.coefSign =        prime256v1_A_sign;
  in.coefA =           prime256v1_absA;
  in.modulus =         prime256v1_Prime;
  in.pointX =          prime256v1_GeneratorX;
  in.pointY =          prime256v1_GeneratorY;
  
  in.scalarMulSize =   ecdsa_priv_prime256v1_priv_len;
  in.scalarMul =       ecdsa_priv_prime256v1_priv;
  
  /* Start the ECC scalar multiplication */
  if(HAL_PKA_ECCMul(&hpka, &in, 5000) != HAL_OK)
  {
    Error_Handler();
  }

  /* Copy the results to user specified space */
  out.ptX = malloc(prime256v1_Prime_len);
  out.ptY = malloc(prime256v1_Prime_len);
  HAL_PKA_ECCMul_GetResult(&hpka , &out);
  /* USER CODE END 2 */
```

The use of HAL_PKA_ECCMulFastMode compared to HAL_PKA_ECCMul requires two separated steps:
- Computation of the montgomery parameter. This parameter can be saved to be reused later (This is the step permitting to save execution time).
- Computation of the ECC multiplication.

The adaptation steps to move from HAL_PKA_ECCMul to HAL_PKA_ECCMulFastMode are:

- Adapt the types of the variables and create a buffer to store the intermediate step.
- Add the additionnal required input parameter of HAL_PKA_ECCMulFastMode (montgomery parameter).
- The result of both operation is the same.

This give the following code:

```
/* USER CODE BEGIN PV */
uint32_t montgomeryparam[8] = {0}; /* Ensure to have enough space for curves not aligned on 4 bytes. In this case: prime256v1_Prime_len/4 */
PKA_HandleTypeDef hpka;
PKA_MontgomeryParamInTypeDef montgomeryparamin = {0};
PKA_ECCMulFastModeInTypeDef in = {0};
PKA_ECCMulOutTypeDef out = {0};
__IO uint32_t operationComplete = 0;
/* USER CODE END PV */
```

```
 /* USER CODE BEGIN 2 */
  
  /* Compute the Montgomery parameter */
  montgomeryparamin.size = prime256v1_Prime_len;
  montgomeryparamin.pOp1 = prime256v1_Prime;
  HAL_PKA_MontgomeryParam(&hpka, &montgomeryparamin, 5000);

  /* Retrieive the result. Result is provided as an array of uint32_t */
  HAL_PKA_MontgomeryParam_GetResult(&hpka, montgomeryparam);

   /* Set input parameters */
  in.modulusSize =     prime256v1_Prime_len;
  in.coefSign =        prime256v1_A_sign;
  in.coefA =           prime256v1_absA;
  in.modulus =         prime256v1_Prime;
  in.pointX =          prime256v1_GeneratorX;
  in.pointY =          prime256v1_GeneratorY;
  
  in.scalarMulSize =   ecdsa_priv_prime256v1_priv_len;
  in.scalarMul =       ecdsa_priv_prime256v1_priv;
  in.pMontgomeryParam = montgomeryparam;
  
  /* Start the ECC scalar multiplication */
  if(HAL_PKA_ECCMulFastMode(&hpka, &in, 5000) != HAL_OK)
  {
    Error_Handler();
  }

  /* Copy the results to user specified space */
  out.ptX = malloc(prime256v1_Prime_len);
  out.ptY = malloc(prime256v1_Prime_len);
  HAL_PKA_ECCMul_GetResult(&hpka , &out);
  /* USER CODE END 2 */
```

Best regards.
Pierre.

dmazzella, 2019-11-27:
@PierreLeCorre 
Thanks for the detailed explanations.
I tested the changes and everything seems to work.

However, I tested the execution of the `HAL_PKA_ECCMulFastMode` with the parameters previously calculated and reused, as you suggested, but I have not seen any improvement in time with respect to `HAL_PKA_ECCMul`.

PierreLeCorre, 2019-11-27:
The difference in execution time is documented inside [RM0434](https://www.st.com/content/ccc/resource/technical/document/reference_manual/group0/83/cf/94/7a/35/a9/43/58/DM00318631/files/DM00318631.pdf/jcr:content/translations/en.DM00318631.pdf)

Refer to Table145 . ECC scalar multiplication
[image]

You need to instrument your code in a manner to be able to observe the difference.

To be the closest possible of those values, you can measure the execution time inside `PKA_Process` function of stm32wbxx_hal_pka.c file with something similar to: 

    start = HAL_GetTick();
    
    /* Start the computation */
    hpka->Instance->CR |= PKA_CR_START;

    /* Wait for the end of operation or timeout */
    if (PKA_PollEndOfOperation(hpka, Timeout, tickstart) != HAL_OK)
    {
      /* Abort any ongoing operation */
      CLEAR_BIT(hpka->Instance->CR, PKA_CR_EN);

      hpka->ErrorCode |= HAL_PKA_ERROR_TIMEOUT;

      /* Make ready for the next operation */
      SET_BIT(hpka->Instance->CR, PKA_CR_EN);
    }
    end = HAL_GetTick();
    result = end-start;

You can also note that this execution time is in cycle not in ms. The example is running @64Mhz. You can observe more easily by reducing the clock speed. The easiest way is to comment the function:

`//SystemClock_Config();`

Like this the example is running at default speed 4Mhz.

dmazzella, 2019-12-02:
@PierreLeCorre @ALABSTM 
Thanks for the detailed and useful explanations.

closed 2019-12-02 by dmazzella