STMicroelectronics/STM32CubeWB #2: USB + BLE (not BLE HID)

pauleffect90, 2020-01-25:
First of all, great job on both the product and this repository. Really well structured and easy to follow.

I'm working on a prototype for a keyboard that also requires wireless connectivity (separated from the HID layer).
I'm basically trying to translate your TransparentModeVCP into a TransparentModeHID but I kinda hit a bump.
Can't really seem to figure out where the initialization of the USB device takes place (there are a ton of includes from different projects / drivers/ middlewares).

Would you kindly point me in the right direction?

ALABSTM, 2020-02-03:
Hi PaulEffect90,

Really glad to read your message! Indeed, our teams working on the WB products did a great job and would certainly appreciate your feedback.

As your question is about both USB and BLE, please allow me to introduce @bouattay and @PierreLeCorre who will answer your questions. Hopefully, you get from them the answers you are looking for.

With regards,

bouattay, 2020-02-05:
Hello PaulEffect90,

Configuring USB using STM32Cube is quite simple, as a starting point can I suggest you to check the provided USB device HID_Standalone application available under this path
Firmware\Projects\P-NUCLEO-WB55.USBDongle\Applications\USB_Device\HID_Standalone

The required files/APIs to build an USB HID  device is available within this application
Please note that USB Class HID supports by default only mouse, you need to customize it by adding required configuration descriptors for a keyboard use case.

Regards,
Aymen

ALABSTM, 2020-02-07:
Hi PaulEffect90,

I hope you are doing well. Any update about your question? Could you find the answers you need thanks to @bouattay's reply? 

With regards,

ALABSTM, 2020-03-04:
Hi PaulEffect90,

Any update? Were the answers of @bouattay's satisfying? Thank you in advance for you reply.

With regards,

ALABSTM, 2020-03-09:
Hi PaulEffect90,

Please allow me to close this issue as we received no feedback from your side since more than a month. Do not hesitate to reopen it in case the point described is still relevant to you. Thank you for your comprehension.

With regards,

closed 2020-03-09 by ALABSTM