STMicroelectronics/STM32CubeWB #3: Example PKA: curve25519

dmazzella, 2020-03-04:
Is it possible to have an example that uses Curve25519 instead P-256 for sign e verify?
Thanks

ALABSTM, 2020-03-10:
Hi Damiano,

Welcome back. I let @PierreLeCorre answer your question.

With regards,

ALABSTM, 2020-04-02:
Hi Damiano,

I hope you are doing well. After discussing your request with our development teams, they said we cannot provide examples with all curves possible.

The [PKA example](https://github.com/STMicroelectronics/STM32CubeWB/tree/master/Projects/NUCLEO-WB35CE/Examples/PKA/PKA_ECDSA_Sign) already provided uses the P-256 elliptic curve as it is one of the most known. You can start from that one and replace the existing curve parameters with the ones corresponding to the Curve25519.

We would have liked to be able to address your request. Thank you for your comprehension. Take care and stay safe.

With regards,

dmazzella, 2020-04-03:
Thanks for the support.

Sorry if it may seem controversial but I have not asked for an example for all possible curves but for one in particular Curve25519.

The request was specific to an example of signature and verification implementation for Curve25519, an example that could demonstrate the EdDSA and non-ECDSA implementation.

But thanks anyway for the support.
Take care of yourself and stay safe too.