STMicroelectronics/STM32CubeWB #6: Error upgrading wireless coprocessor firmware on Nucleo-68

anddam, 2020-06-25:
**Describe the set-up**
Host OS: Linux glibc system
Board: P-NUCLEO-WB55 STM32WB55RG (Nucleo-68)
Tools:
 - STM32CubeProgrammer v2.4.0
 - SMT32CubeWB firmware images

**Describe the bug**
Trying to update the wireless coprocessor to latest v1.7.0 results in an error:
```
Error: failed to download Segment[0]
Error: failed to download the File
Error: Failed to download image!
```

**How To Reproduce**
1. clone `https://github.com/STMicroelectronics/STM32CubeWB`
2. enter the repository
3. checkout tag `v1.7.0`
4. cd `Projects/STM32WB_Copro_Wireless_Binaries/STM32WB5x/`
5. follow instructions at `How to flash the Wireless Coprocessor Binary via USB (Command Line Interface)` in file `Release_Notes.html` using STM32CubeProgrammer v2.4.0

**Additional context**
The *Update History* section of  `Release_Notes.html` shows 1.6.0 changes as latest entry even on tag `v.1.7.0`. However checking out the repo at `v1.6.0` yields the same error result as `v1.7.0`.

**Log**

[bug-ST-P-NUCLEO-WB55.log](https://github.com/STMicroelectronics/STM32CubeWB/files/4830602/bug-ST-P-NUCLEO-WB55.log)

ASELSTM, 2020-06-26:
Hi @anddam,

Thank you for this report. It will be forwarded to our development teams for analysis. We will be back to you as soon as we get feedback.

Thank you for your patience and thank you again for your contribution.

With regards,

ASELSTM, 2020-07-30:
ST Internal Reference: 89179

anddam, 2020-07-30:
> ST Internal Reference: 89179

That's interesting, but can you shed some light on what the actual issue is?

ASELSTM, 2020-08-17:
Hi @anddam,

The issue is currently being handled by our development team. We will get back to you once it is treated. Stay tuned.

With regards

ASELSTM, 2020-11-02:
Hi @anddam, 

According to our development team the problem is related to the CubeProgrammer tool. In fact, the error message generated in the log file shows that the FUS upgrade didn't even started. The fail was at the level of download/erase phase.  

The issue has been solved within the latest version **V2.5.0** of Cubeprogrammer.

Please allow me then to close this thread. Thank you again for your contribution.

With regards,

closed 2020-11-02 by ASELSTM