STMicroelectronics/STM32CubeWB #9: LED1_GPIO_CLK_DISABLE is defined as CLK_ENABLE

HaukeRa, 2020-07-30:
https://github.com/STMicroelectronics/STM32CubeWB/blob/24ba9f167a82143a0a4fa8772cd4466070bcdb82/Drivers/BSP/P-NUCLEO-WB55.Nucleo/stm32wbxx_nucleo.h#L101

Is this for some reason intentional or is it a copy/paste error?

PierreLeCorre, 2020-07-31:
Hello @HaukeRa , effectively this is a copy/paste error. This will be corrected in next release. 
You can note that using the macro will disable the entire port. You should do this only if you have no other peripherals in use on this port.
BR.
Pierre.

ASELSTM, 2020-08-03:
ST Internal Reference: 90673

ASELSTM, 2020-12-09:
Hi @HaukeRa,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.10.0** of the STM32CubeWB published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2020-12-09 by ASELSTM