STMicroelectronics/STM32CubeWB #10: Firmware BLE + 802.15.4

Quadrifiliar, 2020-08-08:
### Feature Request

Could you provide a firmware image (for STM32WB5x)  combining the BLE stack and the MAC 802.15.4 functionality?

ASELSTM, 2020-08-18:
Hi @Quadrifiliar,

Your request has been forwarded to our development teams. We will get back to you as soon as they provide us with their feedback.

With regards,

ASELSTM, 2020-09-24:
Hi @Quadrifiliar, 

We have reported your request to our development team. The firmware image combining the BLE stack and the MAC 802.15.4 is not yet planned for the moment. We will keep you informed if there's any updates.

Please allow me then to close this thread. Thank you for your comprehension and thank you again for your contribution.

With regards,