STMicroelectronics/STM32CubeWB #11: P-NUCLEO-WB55.Nucleo\Examples\ADC\ADC_SingleConversion_TriggerTimer_DMA is fake

LockeKK, 2020-09-17:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.

**Describe the bug**
Like the title, this example is fake. It is supposed to use TIM2 triggering ADC conversion via DMA, but the truth is this is a ADC software trigger ADC DMA continuous conversion.
They are very much the different. If you remove TIM2 totally, ADC DMA is still running.
So this example is totally fake. Yet this example is very important as many users need A timer to trigger ADC in certain period.
Please correct this example. Thanks.

**How To Reproduce**
1. Indicate the global behavior of your application project.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).

3. The use case that generates the problem.

4. How we can reproduce the problem.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

ASELSTM, 2020-09-18:
Hi @LockeKK,

In order to allow a better understanding of the problem, could you please give us more details about the issue and how to reproduce it, so that our development teams can give you an answer.

With regards,

LockeKK, 2020-09-19:
This example intents to use HW timer TIM2 triggers ADC in DMA mode.
But in fact, the ADC is triggered by Software rather then TIM2's TRGO update event. You can find this in the ADC configures.
You can totally remove TIM2, ADC is still working.
Just run this project you will find how to reproduce it.

ASELSTM, 2020-09-21:
Hi @LockeKK,

Thank you for your contribution.

The issue has been confirmed and an bug tracker has been opened internally. A fix will be implemented and made available in the frame of a future release. Thank you once more.

With regards,

ASELSTM, 2020-09-21:
ST Internal Reference: 93065

LockeKK, 2020-09-21:
Good, thanks for your support.
I will close this ticket.

ASELSTM, 2020-09-25:
Reopened till the fix is released on GitHub.

ASELSTM, 2020-12-09:
Hi @LockeKK, 

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.10.0** of the STM32CubeWB published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2020-12-09 by ASELSTM

LockeKK, 2020-12-09:
> Hi @LockeKK,
> 
> I hope you are fine. The issue you reported has been fixed in the frame of version **v1.10.0** of the STM32CubeWB published recently on GitHub.
> 
> Thank you again for having reported.
> 
> With regards,

Thanks for your support.