STMicroelectronics/STM32CubeWB #12: RSSI integer type mismatches from documentation to implementation

tim-nordell-nimbelink, 2020-09-29:
Hello!

In this section of code:
https://github.com/STMicroelectronics/STM32CubeWB/blob/24ba9f167a82143a0a4fa8772cd4466070bcdb82/Middlewares/ST/STM32_WPAN/ble/core/auto/ble_types.h#L237-L244

The documentation states it's a signed integer, but the structure implements an unsigned integer.  This is repeated a few times below in this same source file.

RKOUSTM, 2020-10-04:
Hi @tim-nordell-nimbelink,

Thank you for your contribution. In fact, all parameters are set to unsigned as they are bytes or words in a packet but some of them represent a signed value; it is then up to the application to cast them to the right signed type. Indeed, the value stored in the uint8_t should be interpreted as a signed value in the structure implements. So, we can use it in the code as following int8_t my_rssi = (int8_t)packet->RSSI;

Thank you again for your contribution.

With regards,

tim-nordell-nimbelink, 2020-10-05:
Hi @RKOUSTM -

What about these usages/examples of RSSI in the code?

https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Middlewares/ST/STM32_WPAN/ble/core/auto/ble_hci_le.c#L291-L315

https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Middlewares/ST/STM32_WPAN/ble/core/auto/ble_events.c#L482-L496

And the examples of how to read it out:

https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Projects/P-NUCLEO-WB55.Nucleo/Applications/BLE/BLE_CableReplacement/STM32_WPAN/App/app_ble.c#L492-L496

(This is duplicated in 4 other example projects, too.)

It's error prone to expect every user of a struct to typecast a variable within it, but I'll keep in mind the stance of this codebase as I interact with it further.

-- Tim

RKOUSTM, 2020-10-08:
Hi @tim-nordell-nimbelink,

Thank you for your contribution. Indeed, a fix for this comment below :
https://github.com/STMicroelectronics/STM32CubeWB/blob/e3d0473c1588f1379ff0e4d7d1c78d081c352518/Projects/P-NUCLEO-WB55.Nucleo/Applications/BLE/BLE_CableReplacement/STM32_WPAN/App/app_ble.c#L492-L496

will be updated and made available in future release looks like the following:
```diff
+          /* WARNING: be careful when decoding advertising report as its raw format cannot be mapped on a C structure. 
+          The data and RSSI values could not be directly decoded from the RAM using the data and RSSI field from +hci_le_advertising_report_event_rp0 structure.
+          Instead they must be read by using offsets (please refer to BLE specification).
-          RSSI = *(uint8_t*) (adv_report_data + le_advertising_event->Advertising_Report[0].Length_Data);
+          RSSI = (int8_t)*(uint8_t*) (adv_report_data + le_advertising_event->Advertising_Report[0].Length_Data); 
+          */
```
Thank you once more for this contribution.

With regards,

RKOUSTM, 2020-10-08:
ST Internal Reference: 94206

RKOUSTM, 2021-03-04:
Hi @tim-nordell-nimbelink,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.11.0 of the STM32CubeWB published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-04 by RKOUSTM