STMicroelectronics/STM32CubeWB #13: Missing C++ extern in hci_tl.h

tim-nordell-nimbelink, 2020-10-05:
The file Middlewares/ST/STM32_WPAN/interface/patterns/ble_thread/tl/hci_tl.h is missing C++ externs.  E.g.:

```
#ifdef __cplusplus
extern "C" {
#endif
```

ASELSTM, 2020-10-08:
Hi @tim-nordell-nimbelink,

Thank you for this report. You request has been forwarded to our development teams. We will get back to you as soon as they provide us with their feedback.

With regards,

ASELSTM, 2020-10-08:
Hi @tim-nordell-nimbelink,

The issue has been reported to our development teams and they confirmed it. A fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-10-08:
ST Internal Reference: 94219

ASELSTM, 2020-12-09:
Hi @tim-nordell-nimbelink,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.10.0** of the STM32CubeWB published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2020-12-09 by ASELSTM