STMicroelectronics/STM32CubeWB #14: svc_ctl.c is missing a definition of __weak

tim-nordell-nimbelink, 2020-10-05:
The svc_ctl.c is depending on a compiler specified include to provide the __weak definition.  The .c file should include either  #include "stm32wbxx_hal_def.h", *or* using the CMSIS provided __WEAK definition via "cmsis_compiler.h".  Without modifying this file (or the compiler include directives), you end up with:

```
     Middlewares/ST/STM32_WPAN/ble/svc/Src/svc_ctl.c:71:7: error: expected ';' before 'void'
       71 | __weak void DIS_Init( void )
          |       ^~~~~
          |       ;
    Middlewares/ST/STM32_WPAN/ble/svc/Src/svc_ctl.c:75:7: error: expected ';' before 'void'
       75 | __weak void EDS_STM_Init( void )
          |       ^~~~~
          |       ;
...
```

ASELSTM, 2020-10-08:
Hi @tim-nordell-nimbelink,

Thank you for this other report. Your request has been forwarded to our development team. We will get back to you as soon as they provide us with their feedback.

With regards,

ASELSTM, 2020-10-08:
Hi @tim-nordell-nimbelink,

The issue has been reported to our development team and they confirmed it. A fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-10-08:
ST Internal Reference: 94220

ASELSTM, 2020-12-09:
Hi @tim-nordell-nimbelink,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.10.0** of the STM32CubeWB published recently on GitHub.

Thank you again for having reported.

With regards,

tim-nordell-nimbelink, 2020-12-09:
Hi @ASELSTM -

Unfortunately, the fix isn't quite correct.  The code was changed to use "cmsis_compiler.h", but that provides a __WEAK definition, not a __weak definition.  The rest of the code in the .c file was left with __weak.  The end result is this:

```
Middlewares/ST/STM32_WPAN/ble/svc/Src/svc_ctl.c:64:7: error: expected ';' before 'void'
   64 | __weak void BLS_Init( void )
      |       ^~~~~
      |       ;
Middlewares/ST/STM32_WPAN/ble/svc/Src/svc_ctl.c:68:7: error: expected ';' before 'void'
   68 | __weak void CRS_STM_Init( void )
      |       ^~~~~
      |       ;
Middlewares/ST/STM32_WPAN/ble/svc/Src/svc_ctl.c:72:7: error: expected ';' before 'void'
   72 | __weak void DIS_Init( void )
      |       ^~~~~
      |       ;
Middlewares/ST/STM32_WPAN/ble/svc/Src/svc_ctl.c:76:7: error: expected ';' before 'void'
   76 | __weak void EDS_STM_Init( void )
      |       ^~~~~
      |       ;
...
```

Changing the definitions to __WEAK fixes it:
```
-__weak void BLS_Init( void )
+__WEAK void BLS_Init( void )
...
-__weak void CRS_STM_Init( void )
+__WEAK void CRS_STM_Init( void )
...
```

since that matches the include file that was added.

Thanks,
Tim

ASELSTM, 2020-12-10:
Hi @tim-nordell-nimbelink,

Thank you for bringing this to our attention, I will let our development team know about this. Thank you once again for your contribution.

Best regards,

ASELSTM, 2022-01-13:
ST Internal Reference: 120815.

karlp, 2022-01-24:
Please definitely change the __weak to __WEAK so that things "just work"  please definitely _don't_ add a HAL dependency to this core library :)

tim-nordell-nimbelink, 2022-04-27:
This looks like this was fixed in v1.13.2.