STMicroelectronics/STM32CubeWB #15: Binaries missing and project build error

rkal01, 2020-10-08:
There is no binary directory in the following project (as mentioned in the readme file)  :
STM32CubeWB/Projects/P-NUCLEO-WB55.Nucleo/Applications/BLE/BLE_Custom/

Not able to build the project via STM32CubeIDE (v1.4.2), even if the ioc. file is configured and code generated

[image]

ASELSTM, 2020-11-02:
Hi @rkal01,

Our development team has confirmed your issue. In fact, erroneous comments have been inserted in the readme file. The file will then be updated as following : 

```diff
In order to make the program work, you must do the following :
 - Open your preferred toolchain 
 - Rebuild all files and load the image into Target memory
-  - Or use the BLE_Custom.hex from Binary directory
 - This must be done for BLE_Custom (MB1355C) 
```
 The fix will be published in the frame of a future publication. We cannot share a date for the moment. Stay tuned and thank you once more for your contribution and for your patience.

With regards,

ASELSTM, 2020-11-02:
ST Internal Reference: 95336

ASELSTM, 2020-12-09:
Hi @rkal01,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.10.0** of the STM32CubeWB published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2020-12-09 by ASELSTM