STMicroelectronics/STM32CubeWB #17: LPUART1 of Mac_802_15_4_FFD project doesn't work.

ty4tw, 2020-10-09:
In  Mac_15_4_FFD/Cor/Inc/hw_conf.h   should be changed as follows:

````
/*
#define CFG_HW_LPUART1_TX_PORT_CLK_ENABLE      __HAL_RCC_GPIOA_CLK_ENABLE
#define CFG_HW_LPUART1_TX_PORT                 GPIOA
#define CFG_HW_LPUART1_TX_PIN                  GPIO_PIN_2
*/
#define CFG_HW_LPUART1_TX_PORT_CLK_ENABLE      __HAL_RCC_GPIOC_CLK_ENABLE
#define CFG_HW_LPUART1_TX_PORT                 GPIOC
#define CFG_HW_LPUART1_TX_PIN                  GPIO_PIN_1
````
````
/*
#define CFG_HW_LPUART1_RX_PORT_CLK_ENABLE      __HAL_RCC_GPIOA_CLK_ENABLE
#define CFG_HW_LPUART1_RX_PORT                 GPIOA
#define CFG_HW_LPUART1_RX_PIN                  GPIO_PIN_3
*/
#define CFG_HW_LPUART1_RX_PORT_CLK_ENABLE      __HAL_RCC_GPIOC_CLK_ENABLE
#define CFG_HW_LPUART1_RX_PORT                 GPIOC
#define CFG_HW_LPUART1_RX_PIN                  GPIO_PIN_0
````

ASELSTM, 2020-10-19:
Hi @ty4tw,

Tank you for this new issue you have raised out.

In order to allow a better understanding of the problem, could you please give us more details about the issue. Would you also describe how you did to detect this issue and how to reproduce it, so that our development teams can give you an answer.

With regards,

ty4tw, 2020-10-20:
Hi @ASELSTM 

 I debugged the application and found the bug and fixed it.

How to detect:
Connect USB serial module to PC0 and PC1 pins.  Those are LPUART1 RX-TX ports.
Build ***Mac_802_15_4_FFD project*** and run with ***STM32CubeIDE***.   
Nothing shown on the terminal screen. Some messages should be shown.

what I fixed:
Change port from ***GPIOA*** to ***GPIOC*** in ***Mac_15_4_FFD/Cor/Inc/hw_conf.h*** file.
Change port No from ***GPIO_PIN_2, GPIO_PIN_1*** to ***GPIO_PIN_3, GPIO_PIN_0***.

Result:
I got messages on the terminal screen.

Thank you but I don't need any answer. I just reported instead of PR.

ty4tw, 2020-10-20:
In readme,txt
```
 if you want to get the traces in real time, you can connects an HyperTerminal 
 via an UART RS232 cable using the following connections :
  RXD : CN10(Pin35)   
  TXD : CN10(Pin37)
```
Reason why I did not follow the readme.txt is
line 108 of app_conf.h defines CFG_DEBUG_TRACE_UART as follows:
```
/******************************************************************************
 * UART interfaces
 ******************************************************************************/

#define CFG_DEBUG_TRACE_UART      hw_lpuart1
#define CFG_CLI_UART                hw_uart1
```

ASELSTM, 2020-10-22:
Hi @ty4tw,

Thank you for the details. 

After checking, the debug trace is actually working with **PA2**(Pin 35 CN10)  and **PA3**(Pin 37 CN10) pins as described in the readme file. In fact, PA2 and PA3 are also LPUART1 RX-TX ports.

Please allow me then to close this issue. You can reopen it if you have anything else relevant to add regarding this topic.

With regards,

closed 2020-10-22 by ASELSTM