STMicroelectronics/STM32CubeWB #18: STM32CubeIDE  Compile error of Mac_802_15_4_FFD

ty4tw, 2020-10-10:
${PARENT-1-PROJECT_LOC}/STM32_WPAN/***app***/app_ffd_mac_802_15_4_process.c
${PARENT-1-PROJECT_LOC}/STM32_WPAN/***app***/app_ffd_mac_802_15_4.c

Change app to ***App*** to avoid compile error.

[image]

ASELSTM, 2020-11-03:
Hi @ty4tw,

The issue has been confirmed and an bug tracker has been opened internally. A fix will be implemented and made available in the frame of a future release. Thank you once more for your contribution.

With regards,

ASELSTM, 2020-11-03:
ST Internal Reference: 95353

ASELSTM, 2020-12-09:
Hi @ty4tw,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.10.0** of the STM32CubeWB published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2020-12-09 by ASELSTM