STMicroelectronics/STM32CubeWB #20: BLE_Hid application example missing .ioc file

Daniel-Khodabakhsh, 2020-11-14:
The `Projects/P-NUCLEO-WB55.Nucleo/Applications/BLE/BLE_Hid` application example appears to be missing an `.ioc` file.
I've tried using the `STM32CubeIDE` to create a new project based on this application example and that doesn't generate the `.ioc` file either.

Is there an `.ioc` file I can borrow from another BLE project which would work with `BLE_Hid`? Or is there a step I'm missing to generate this file?

ASELSTM, 2020-11-18:
Hi @Daniel-Khodabakhsh,

Thank you for this report. Your question has been forwarded to our development teams. We will get back to you as soon as they provide us with their feedback.

With regards,

Daniel-Khodabakhsh, 2020-12-22:
Closing, saw the application note AN5155 which states that BLE_Hid is not an MX project.

closed 2020-12-22 by Daniel-Khodabakhsh