STMicroelectronics/STM32CubeWB #24: How to use STM32WB BLE event callbacks?

Daniel-Khodabakhsh, 2020-12-22:
The STM32CubeWB libraries contain a file which specifies the BLE event callback functions: [`STM32CubeWB/Middlewares/ST/STM32_WPAN/ble/core/auto/ble_events.c`](https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Middlewares/ST/STM32_WPAN/ble/core/auto/ble_events.c) which is great: I'd rather use this than the switch/case statements.

However when I try to use the callbacks, they don't work.

I've tried adding a simple trace to see if the event ever gets triggered but I don't see any messages (both with and without the `__WEAK` attribute):
```c
__WEAK void hci_le_connection_complete_event(
	uint8_t Status,
	uint16_t Connection_Handle,
	uint8_t Role,
	uint8_t Peer_Address_Type,
	const uint8_t* Peer_Address,
	uint16_t Conn_Interval,
	uint16_t Conn_Latency,
	uint16_t Supervision_Timeout,
	uint8_t Master_Clock_Accuracy
)
{
	Debug_print("hci_le_connection_complete_event called!");
}
```

But in my switch/case events I do get the `EVT_LE_CONN_COMPLETE` event triggering.

Is there additional (or alternative) setup I need to perform? Or does it only work with a specific version of the BLE binaries loaded onto the co-processor?

RKOUSTM, 2020-12-22:
Hi @Daniel-Khodabakhsh,

Thank you for you contribution. In order to allow a better analysis of this problem, could you please give us more details about the application that you are used to reproduce this issue, so that our development teams can give you an answer. 
Could you please confirm you are using the latest firmware version and that the issue is systematically reproducible on your side?

Thank you again for your contribution.

With regards,

Daniel-Khodabakhsh, 2020-12-22:
Hey @RKOUSTM, sure thing!

I'm using the latest firmware available today: `v1.10.0`
The firmware version loaded via STM32CubeProgrammer is: `V2J37M26`
Also want to mention that I'm using the `P-NUCLEO-WB55` development board.

Will add this information in the original comment above as well.

RKOUSTM, 2021-01-04:
Hi @Daniel-Khodabakhsh,

Thank you for this report. To reproduce the issue you described, did you used one of our ``ble`` applications ? Otherwise, it is possible to share with us a code snippet allowing our development teams to reproduce it.  

Thank you again for your contribution.

With regards,

Daniel-Khodabakhsh, 2021-01-05:
Hello @RKOUSTM,

My pleasure =) thanks for following up.

Yes I'm using one of your `ble` applications: `Server profile > Custom Template`. I've attached a toy project to this post which demonstrates the issue. In this toy project, the only code I added was code within the user sections which don't get clobbered by the ioc code generator.

Here are the steps to reproduce the issue, assuming my project is used. You can also achieve the same results by creating a new project from scratch for the `P-NUCLEO-STM32WB` development board with the `STM32_WPAN` middleware set to `BLE` and using the `Custom Template` server profile/mode.

[image]

### Steps to reproduce:

1. Open my attached project with STM32CubeIDE.

2. Build and run the project on the development board.

3. Connect to the USART1 serial output from the onboard ST-Link.

4. Use a phone to find the BLE device named `test-7` and connect to it, supplying `111111` as the pin.

5. Observe the serial output displays the following message:
```
EVT_LE_CONN_COMPLETE triggered via switch/case!
```

### Problem:

The output should also contain the message below but it does not.
```
hci_le_connection_complete_event called!
```

I have an implementation of `hci_le_connection_complete_event` in `/STM32_WPAN/App/app_ble.c:936` which to my understanding from the documentation, should be triggered when the BLE connection completes.

However there appears to be some missing glue related to the contents of `/Middlewares/ST/STM32_WPAN/ble/core/auto/ble_events.h`.

[STM32WB Firmware Shadow.zip](https://github.com/STMicroelectronics/STM32CubeWB/files/5767584/STM32WB.Firmware.Shadow.zip)

RKOUSTM, 2021-01-12:
Hi @Daniel-Khodabakhsh,

Thank you for your report. Indeed, the function ``__WEAK void hci_le_connection_complete_event()`` in the ``app_ble.c`` file is never called, so it will never be executed. 

To simply display a message about the ``EVT_LE_CONN_COMPLETE`` event, the above function is useless, if we look at the ``case EVT_LE_CONN_COMPLETE`` in the ``SVCCTL_App_Notification()`` function, the call of the ``Custom_APP_Notification()`` function raises the event on the application side in the``custom_app.c`` file.  In this function, you just have to add the message in the ``case CUSTOM_CONN_HANDLE_EVT``.  Then, in the ``case EVT_LE_CONN_COMPLETE`` of the ``SVCCTL_App_Notification()`` function in the ``app_ble.c`` file, the message ``EVT_LE_CONN_COMPLETE triggered via switch/case!`` is already displayed in the terminal. 

According to our development team, you are redeveloping the HID service that has already been developed in the Middleware (``hids.c``). So, you can develop your application part based on this hids.c middleware. Also, you can take inspired from the [BLE_Hid](https://github.com/STMicroelectronics/STM32CubeWB/tree/master/Projects/P-NUCLEO-WB55.Nucleo/Applications/BLE/BLE_Hid) application proposed by ST on P-NUCLEO6WB55 Nucleo with different IDEs (IAR, MDK-ARM and STM32CubeIDE) that all of them are working well.

Please let us know your feedback so that we can help you. Thank you again for your contribution. 

With regards,

Daniel-Khodabakhsh, 2021-01-12:
Hey @RKOUSTM,

The project I uploaded isn't derived from BLE_Hid in any way. This project I uploaded is using USB HID (no BLE at all).

BLE portions of the project are generated purely from the ioc project generator (STM32CubeMX). All of what you said about the switch case statements are true, but all of this is generated via the STM32CubeMX, I didn't write any of this myself. As I said in my last post, **the only modifications I did to the BLE was writing within the user sections**.

> ... the only code I added was code within the user sections which don't get clobbered by the ioc code generator

None of the application templates use the STM32WB BLE event callbacks at all, I checked them all.

This is an issue because the STM32WB BLE event callbacks are mentioned in the documentation.

So a solution to this would either to be update the documentation, or add an actual example of the STM32WB BLE event callbacks being used in one of the application templates.

If it's the case that STM32WB BLE event callbacks are used, what are the steps I can take, via STM32CubeIDE or STM32CubeMX to see them?

RKOUSTM, 2021-01-18:
Hi @Daniel-Khodabakhsh,

Thank you for your report. However, as this is related to **Cube MX**, it is out of our scope. In general terms, we don't treat aspects related to CubeMX tool at GitHub level, they are rather treated at [ST Community](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) level. There you shall find people who will address your request and whom you will hopefully find a solution with.
Now, as this issue is not directly related to software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to ecosystem (CubeMX tool), please allow me to close it.

Thank you for your comprehension and thank you again for your contribution. 

With regards,

RKOUSTM, 2021-01-18:
Hi @Daniel-Khodabakhsh,

After further discussion with our development team, we will keep this reported point open and see if there are plans to add a BLE application using ``event callbacks`` in the frame of a future release. We will get back to you with explanation on how to use it in the development of your application as soon as we have more details.

Will keep you informed.
With regards,

RKOUSTM, 2021-03-24:
Hi @Daniel-Khodabakhsh,

I hope you are doing well. A BLE application with event callbacks will be made available in the frame of the future release `v1.12.0`. Thank you once again for your contribution.

With regards,

Daniel-Khodabakhsh, 2021-03-26:
Thanks @RKOUSTM! Really appreciate it 👍

RKOUSTM, 2021-08-16:
Hi @Daniel-Khodabakhsh,
 
 New BLE applications are now available in the frame of the STM32CubeWB FW package V1.12.0 release. 

This issue can be closed now. Thank you again for your contribution.

With regards,

closed 2021-08-16 by RKOUSTM