STMicroelectronics/STM32CubeWB #27: Generated Advertising Data, Manufacturer Specific doesn't observe Data Type Specification

karlp, 2021-04-09:
**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
   P-Nucleo-WB55.Nucleo (but not relevant)
 * IDE or at least the compiler and its version.
  CUBE IDE 1.6.1 +  Cube WB 1.11.0

**Describe the bug**
The Advertising Data generated uses a manufacturer specific data block as shown below. This is in the generated code section that cannot be edited, in this case, in 
```
static const char local_name[] = { AD_TYPE_COMPLETE_LOCAL_NAME ,'H','R','S','T','M'};
uint8_t  manuf_data[14] = {
    sizeof(manuf_data)-1, AD_TYPE_MANUFACTURER_SPECIFIC_DATA,
    0x01/*SKD version */,
    0x00 /* Generic*/,
    0x00 /* GROUP A Feature  */,
    0x00 /* GROUP A Feature */,
    0x00 /* GROUP B Feature */,
    0x00 /* GROUP B Feature */,
    0x00, /* BLE MAC start -MSB */
    0x00,
    0x00,
    0x00,
    0x00,
    0x00, /* BLE MAC stop */
};
```

This does not observe the format of Advertising type: Manufacturer Specific data, as specified in Supplement to Bluetooth Core Specification | CSS v9, Part A, section 1.4.  

[image]

The AD Length is correct, and the AD type of 0xff is correct, but the next two bytes should be ST's 16 bit company id, 0x0030, and _then_ the ST custom data.

This should also be updated in AN5289 - "Building wireless applications with stm32wb"
This should also be updated in all your examples and demo apps too :)

ASELSTM, 2021-06-08:
Hi @karlp,

Allow me first to thank you for your contribution. Your report has been forwarded to our development team. We will get back to you as soon as we have more details.

With regards,

ASELSTM, 2021-06-08:
ST Internal Reference: 108594

ASELSTM, 2021-07-07:
Hi @karlp,

The BLE examples have been created to be compliant with **Blue ST protocol** which is an ST custom format for Blue android application, for details please refer to [this ](https://github.com/STMicroelectronics/BlueSTSDK_Android)GitHub repository.

In BLE STM32CubeWB package, we do only provide BLE applications and examples running on STM32WB HW but not final SW products. So that, if you need to align with **“manufacturer specific data”** described in CSS_v9 specification, you can change custom `manuf_data[]` structure used to fulfill data part of AD elements by replacing 2 first bytes after AD type field (containing SDK version and generic feature on 2 bytes) by ID company ST number “0x0030”. However, you will loose compliancy with ST BLE sensor android application (cf BlueSTSDK_Android link for details).  

Actually,  manufacturer ID has not been implemented yet. It will be available in the next version of BlueSTSDK. This field indicates the Manufacturer ID of STMicroelectronics which corresponds to 0x0030. This field will have a length of 2 Bytes and will be located after the "Length" field.

With regards,

ASELSTM, 2021-07-08:
Hi @karlp,

Please allow me then to close this thread. If you have any updates you can reopen it at any time. Thank you for your comprehension.

With regards,

closed 2021-07-08 by ASELSTM