STMicroelectronics/STM32CubeWB #29: v1.2.0 stm32wb5x_FUS_fw.bin version is not reported correctly and fails to upgrade

EZIMM, 2021-06-18:
STM32WB55RE (512K)
CubeProgrammer v2.7.0

CubeProgrammer will not upgrade latest STM32WB5x FUS v1.2.0 from latest 1.11.1 pack.
The Selected File version is being shown as **FUS V-1.-1.-1**
If we try to run the upgrade with this file, the log reports

> Could not execute firmware upgrade command, Impossible to install FUS V1.0.1 on 256/512KB Flash Parts

Also tried downloading the 1.11.1 patch software pack from ST website and it has the same problem.

[image]

RKOUSTM, 2021-07-07:
Hi @EZIMM,
 
Thank you for your reports. I have checked the reported problem using STM32WB55RG board and no problem detected for the upgrade of the stm32wb5x_FUS_fw.bin (v1.1.2) version for STM32CubeWB v1.11.1.

I checked this problem using the STM32WB55RG board and no problem was detected for the stm32wb5x_FUS_fw.bin (v1.1.2) update on STM32WB Cube FW V1.11.1.

[image]

Concerning the FUS V1.0.1 installation issue on 256/512KB Flash parts, your request has been forwarded to our development teams to check this regression in the STM32CubeWB from v1.11.0 to v.11.1. We will be back to you as soon as we get an answer.

Thank you again for reports.

With regards,

EZIMM, 2021-07-07:
@RKOUSTM Thanks for the reply.

1. According to the release notes in v1.11.1, the FUS version should be v1.2.0, not v1.1.2 you mentioned. Is that a mistake in the release notes?
[image]
1. I see in your screenshot you are using the "Erasing & Programming" tab of CubeProgrammer to update the FUS, and not the "Firmware Upgrade Services" tab that I showed. Can you confirm if you see the issue in the "Firmware Upgrade Services" tab? You should need to start the FUS prior to upgrading it with the latest release notes instructions.

EZIMM, 2021-07-08:
Update: I have also confirmed that this appears to be an issue with the 256/512K parts, as my new board with the 1M part works fine with the v1.2.0 FUS from the v1.11.1 patch. I can work around this issue until an update is released to CubeProgrammer. Thanks!

RKOUSTM, 2021-07-13:
Hi @EZIMM,

Thank you for reporting this point. Concerning the upgrade of `FUS v1.2.0` from the latest STM32CubeWB v1.11.1 package on STM32WB55RG (1M), no problems are detected with using `CubeProgrammer` :

[image]

Regarding the upgrade problem of the latest FUS version on 256/512K parts, it has ben reported to `CubeProgrammer` development teams who will take in charge the issue. 

An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release of CubeProgrammer.

As this is a CubeProgrammer-related issue and not directly related to the firmware published in this repository, please allow me to close it. 

Thank you for your comprehension and contribution.

With regards,

RKOUSTM, 2021-07-13:
ST Internal Reference: 110531

closed 2021-07-13 by RKOUSTM