STMicroelectronics/STM32CubeWB #31: Issue with STM32WB55 USB Custom Hid using V1.11.1 firmware pack.

matheusmbgeng, 2021-07-02:
Hello guys, how are u? Hope you're all fine =)

I had my application developed using V.10.1 version firmware pack . It was requested to me to update firmware to use V1.11.1 firmware pack. As i upgraded, several things messed up, including CustomHID. 

Here's what is happening:

In CUSTOM_HID_OutEvent_FS function, my device acknowlegde the received package using 

USBD_CUSTOM_HID_SendReport. This was working perfectly with V1.10.1. Now that i upgraded, my whole USB comm gets messed up after the first sent ACK. After the first sent package, my application can't received neither send data via USB . If i remove the ACK send, i'm able to continuous receiving data.

What i noticed after updating:

PCB and USB LL libraries got changed. After plenty debugging, i notice that this specific function was messing all up: PCD_EP_ISR_Handler. 

This function changed from V1.10.1 update to V1.11.1. I studied this function and USB code flow and discovered a change that made my USB comm back to work. This are the last lines of  EP_ISR_Handler function:
```        
if ((ep->type == EP_TYPE_BULK) && ((wEPVal & USB_EP_KIND) == 0U))
        {
          /* multi-packet on the NON control IN endpoint */
          TxByteNbre = (uint16_t)PCD_GET_EP_TX_CNT(hpcd->Instance, ep->num);

          if (ep->xfer_len > TxByteNbre)
          {
            ep->xfer_len -= TxByteNbre;
          }
          else
          {
            ep->xfer_len = 0U;
          }

          /* Zero Length Packet? */
          if (ep->xfer_len == 0U)
          {
            /* TX COMPLETE */
#if (USE_HAL_PCD_REGISTER_CALLBACKS == 1U)
            hpcd->DataInStageCallback(hpcd, ep->num);
#else
            HAL_PCD_DataInStageCallback(hpcd, ep->num);
#endif /* USE_HAL_PCD_REGISTER_CALLBACKS */
          }
          else
          {
            /* Transfer is not yet Done */
            ep->xfer_buff += TxByteNbre;
            ep->xfer_count += TxByteNbre;
            (void)USB_EPStartXfer(hpcd->Instance, ep);
          }
        }
        /* Double Buffer Iso/bulk IN (bulk transfer Len > Ep_Mps) */
        else
        {
          (void)HAL_PCD_EP_DB_Transmit(hpcd, ep, wEPVal);
        }
      }
```

What made my USB comm back to work was changing HAL_PCD_EP_DB_Transmit to HAL_PCD_EP_Transmit in last lines of code snippet above.

I'm far from being a USB expert and being honest, i don't know what are the exactly effects that this function change does to USB flow. What i know is that i shouldn't modified ST Drivers for my application to work.

My custom HID was already working, following tutorial available on ST youtube channel. I'm using usb settings exactly as described in the tutorial for 64 bytes packages receiving/sending.

Things i already tried:

-> Created a project with V1.11.1 only using CUSTOM Hid. Got the same behaviour.

-> Tried to use V1.11.0. Got the same behaviour.

All that explained, does someone could give me a hint of what should i do? I NEED to update the firmware pack to a newest version.

ALABSTM, 2021-09-23:
Hello @matheusmbgeng,

We are fine thank you. We hope you too.

Our development teams confirmed the point you reported. Actually, it has been already spotted internally and the fix is already available in the frame of STM32CubeWB **v1.12.0**. 

We hope that you already upgraded your application to that new version and we do apologize for the somewhat delayed reply.

With regards,

ALABSTM, 2021-09-23:
ST Internal Reference: 104250

closed 2021-09-23 by ALABSTM