STMicroelectronics/STM32CubeWB #36: Minor documentation fix

atsju, 2021-07-16:
```C
/**
 * @brief Initialize the System Host Controller Interface.
 *        This function must be called before any communication on the System Channel
 *
 * @param  pData: System events callback function pointer
 *         This callback is triggered when an user event is received on
 *         the System Channel from CPU2.
 * @param  pConf: Configuration structure pointer
 * @retval None
 */
void shci_init(void(* UserEvtRx)(void* pData), void* pConf);
```
should be
```C
/**
 * @brief Initialize the System Host Controller Interface.
 *        This function must be called before any communication on the System Channel
 *
 * @param  UserEvtRx: System events callback function pointer
 *         This callback is triggered when an user event is received on
 *         the System Channel from CPU2.
 * @param  pConf: Configuration structure pointer
 * @retval None
 */
void shci_init(void(* UserEvtRx)(void* pData), void* pConf);
```

Sorry for many issues but I'm trying to understand how the code works so actually yes: I am reading the documentation.

RKOUSTM, 2021-08-06:
Hi @atsju,

Thank you for your contribution. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-08-06:
ST Internal Reference: 111682

closed 2021-12-08 by RKOUSTM