STMicroelectronics/STM32CubeWB #37: Fix assert_param(IS_RCC_PERIPHCLOCK(PeriphClkInit->PeriphClockSelection));

SergeiSOficial, 2021-08-02:
**Describe the set-up**
 * Custom board STM32WB30CE
 * EWARM 8.30

**Describe the bug**
Need adding RCC_PERIPHCLK_CLK48SEL in assert_param(IS_RCC_PERIPHCLOCK(PeriphClkInit->PeriphClockSelection)), because it used in RNG module

**How To Reproduce**
1. #define USE_FULL_ASSERT

2. in CubeMX add RNG module with clock CLK48 with MSI

RKOUSTM, 2021-08-13:
Hi @SergeiSOficial,

This duplicated issue will remain closed. A further detail of your reported point is answered in Issue [#39](https://github.com/STMicroelectronics/STM32CubeWB/pull/39).

With regards,

RKOUSTM, 2021-08-13:
Duplicate of [#39](https://github.com/STMicroelectronics/STM32CubeWB/pull/39).

closed 2021-08-13 by RKOUSTM