STMicroelectronics/STM32CubeWB #41: Could not program the bin file

Lozeonjyu, 2021-08-09:
Hi,

I am trying to burn the bin file to the microcontroller STM32WB55CGUx. The FUS version is 1.1.1.0. I was trying to update the FUS version but "error:failed to download image" came up. I also tried to program the full stack ble directly and came back with error as well. Please check the log(I use the GUI):

> 16:28:38 : Time elapsed during the read operation is: 00:00:00.001
> 16:28:40 : Download Stack/FUS image at address 0x80ec000 ...
> 16:28:40 : Memory Programming ...
> 16:28:40 : Opening and parsing file: stm32wb5x_FUS_fw.bin
> 16:28:40 :   File          : stm32wb5x_FUS_fw.bin
> 16:28:40 :   Size          : 122134 Bytes
> 16:28:40 :   Address       : 0x080EC000 
> 16:28:40 : Erasing memory corresponding to segment 0:
> 16:28:40 : Operation exceeds memory limits
> 16:28:40 : failed to erase memory
> **16:28:40 : Error: Failed to download image!**  
> 16:29:19 : UPLOADING ...
> 16:29:19 :   Size          : 4 Bytes
> 16:29:19 :   Address       : 0x1FFF8070
> 16:29:19 : Read progress:
> 16:29:19 : Data read successfully
> 16:29:19 : Time elapsed during the read operation is: 00:00:00.001
> 16:29:21 : Download Stack/FUS image at address 0x80cb000 ...
> 16:29:21 : Memory Programming ...
> 16:29:21 : Opening and parsing file: stm32wb5x_BLE_Stack_full_fw (1).bin
> 16:29:21 :   File          : stm32wb5x_BLE_Stack_full_fw (1).bin
> 16:29:21 :   Size          : 122507 Bytes
> 16:29:21 :   Address       : 0x080CB000 
> 16:29:21 : Erasing memory corresponding to segment 0:
> 16:29:21 : Erasing internal memory sectors [203 232]
> 16:29:21 : Download in Progress:
> 16:29:23 : File download complete
> 16:29:23 : Time elapsed during download operation: 00:00:02.851
> 16:29:23 : Verifying ...
> 16:29:23 : Read progress:
> 16:29:24 : Download verified successfully 
> 16:29:24 : Firmware Upgrade process started ...
> 16:29:24 : Application is running, Please Hold on...
> 16:29:36 : Reconnecting...
> 16:29:36 : Reconnected 
> **16:29:36 : Error: FUS_STATE_IMG_NOT_FOUND**

I followed the address table so it should be fine.

Lozeonjyu, 2021-08-09:
It seems like I read the FUS version on 0x20010010.

In the release not file it mentions

> Note: when FUS is not running but FUS operator used, you can read FUS version @0x20010010

What should I do if I have this?

Thanks

RKOUSTM, 2021-08-13:
Hi @Lozeonjyu,

According to the release notes, the FUS version should be v1.2.0 and the install address for the target binary for STM32WB55CGUx board should be 0x080EC000. I have checked the reported problem using the "Firmware Upgrade Services" tab of `CubeProgrammer` to update the FUS on STM32WB55RG (1M) board and no problem is detected for the upgrade of the stm32wb5x_FUS_fw.bin (v1.2.0) version.

[image]

Thank you for your comprehension and contribution.

With regards,

RKOUSTM, 2021-08-13:
Hi @Lozeonjyu,

I tried to reproduce the issue you have described, with no success. Indeed, I have checked the reported problem by using the `Firmware Upgrade Services` tab of the `CubeProgrammer` tools to update the `FUS` on the `STM32WB55RG (1M)` board and no problem is detected for the upgrade of the `stm32wb5x_FUS_fw.bin (v1.2.0) `version. According to the release notes, the `FUS `version should be `v1.2.0` and the target binary installs address for STM32WB55CGUx board you are used should be `0x080EC000`. 

[image]

Regarding the point for which you opened the issue, as it is irrelevant, allow me to close the issue.

Thank you for your comprehension and contribution.

With regards,

closed 2021-08-13 by RKOUSTM