STMicroelectronics/STM32CubeWB #46: warning: braces around scalar initializer

AndreiCherniaev, 2021-11-01:
https://github.com/STMicroelectronics/STM32CubeWB/blob/4a9f9892bfeeb155fcae93a493f3f358f193ee9d/Projects/P-NUCLEO-WB55.USBDongle/Applications/BLE/BLE_MeshLightingLPN/Core/Src/app_debug.c#L68

There are two warnings on one string! It is not normal. 
warning: braces around scalar initializer
warning: excess elements in scalar initializer

ASELSTM, 2021-11-05:
Hi @AndreiCherniaev,

Would you give us details about the compiler you have used detect these warnings ? 

With regards,

ASELSTM, 2021-11-29:
Hi @AndreiCherniaev,

Please allow me to close this thread as no activity. You can reopen it at any time if you have details to provide us to help you solve this issue.
Thank you for your comprehension and thank you once more for your contribution.

With regards,

closed 2021-11-29 by ASELSTM