STMicroelectronics/STM32CubeWB #47: Init_Exti() does not enable interrupt lines

maje91, 2021-12-03:
The function `Init_Exti` found at `Projects/P-NUCLEO-WB55.Nucleo/Applications/BLE/BLE_HeartRate/Core/Src/app_entry.c:163`
is supposed to enable interrupt lines for IPCC and HSEM. The corresponding bits are `&`-ed instead of `|`-ed causing the argument to `LL_EXTI_EnableIT_32_63` to be zero and thus no interrupt lines are enabled. `grep`ing for this shows that the error is present in many of the example applications.

ALABSTM, 2022-01-30:
Hi @maje91,

Thank you for this report. We will get back to you with a feedback as soon as possible. Please excuse the delay it may take us sometimes to reply. Thank you for your comprehension.

With regards,

RKOUSTM, 2022-02-25:
Hi @maje91,

 First of all we would like to thank you for your report. 

An internal tracker has been logged and a fix will be published in the frame of a future release.

Thank you again for your report.

With regards,

RKOUSTM, 2022-02-25:
ST Internal Reference: 123480

ASELSTM, 2022-07-08:
Hi @maje91,

Thank you for your contribution. This issue has been fixed in the frame of version v1.14.0 of the STM32CubeWB published recently. Please allow me then to close this thread.

Thank you again for your contribution.

With regards,

closed 2022-07-08 by ASELSTM