STMicroelectronics/STM32CubeWB #48: stm32wbxx_hal_exti.c fails to build with -Werror=unused-parameter

juliengros, 2021-12-05:
**Describe the bug**
stm32wbxx_hal_exti.c fails to build with -Werror=unused-parameter. 
[image]

**Possible Fix**
The following change fixes the issue. I would be more than happy contributing but unclear to me what CLA I should get with ST to contribute. Maybe you can help me clarifying.
[image]

RKOUSTM, 2021-12-09:
Hi @juliengros,

Thank you for this contribution. The procedure is very simple and is to be done just **once**. Just follow this [link](https://cla.st.com) above, read carefully the terms of the CLA, then, if you accept them, click on the `I Accept` button. Thank you for again for your contribution.

With regards,

RKOUSTM, 2021-12-09:
ST Internal Reference: 119231

RKOUSTM, 2022-01-18:
Hi @juliengros,

Thank you for your report. This duplicated issue will remain closed. The pull-request #50 that you reported will be kept open to track the point.

Thank you for you again for your report.

With regards,

RKOUSTM, 2022-01-18:
Duplicate of #50.

closed 2022-01-18 by RKOUSTM