STMicroelectronics/STM32CubeWB #51: PCD Connect and Disconnect callbacks are not called anywhere in the code

fronders, 2022-02-03:
**Describe the set-up**
 * CubeMX v6.4.0 with CubeWB v1.13.1

**Describe the bug**
Generated code in usbd_conf.c contains `HAL_PCD_ConnectCallback()` and `HAL_PCD_DisconnectCallback()`:
``` c
/**
  * @brief  ConnectCallback callback.
  * @param  hpcd: PCD handle
  * @retval None
  */
void HAL_PCD_ConnectCallback(PCD_HandleTypeDef * hpcd)
{
  USBD_LL_DevConnected(hpcd->pData);
}

/**
  * @brief  Disconnect callback.
  * @param  hpcd: PCD handle
  * @retval None
  */
void HAL_PCD_DisconnectCallback(PCD_HandleTypeDef * hpcd)
{
  USBD_LL_DevDisconnected(hpcd->pData);
}
```

But these callbacks are not called anywhere in stm32wbxx_hal_pcd. Is this a bug, or what if the point of these functions then?

ALABSTM, 2022-04-12:
Hi @fronders,

Would you mind double-checking this point please? Below code snippets show that both callbacks are defined in [stm32wbxx_hal_pcd.c](https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Drivers/STM32WBxx_HAL_Driver/Src/stm32wbxx_hal_pcd.c) since version **1.13.0** of the STM32CubeWB firmware.

With regards,

https://github.com/STMicroelectronics/STM32CubeWB/blob/398d0fbc5ef491da30f777575b911e50e68fa77f/Drivers/STM32WBxx_HAL_Driver/Src/stm32wbxx_hal_pcd.c#L148-L149

https://github.com/STMicroelectronics/STM32CubeWB/blob/398d0fbc5ef491da30f777575b911e50e68fa77f/Drivers/STM32WBxx_HAL_Driver/Src/stm32wbxx_hal_pcd.c#L1295-L1303

https://github.com/STMicroelectronics/STM32CubeWB/blob/398d0fbc5ef491da30f777575b911e50e68fa77f/Drivers/STM32WBxx_HAL_Driver/Src/stm32wbxx_hal_pcd.c#L1310-L1318

fronders, 2022-04-12:
Hello @ALABSTM!

Yes, these functions are assigned to `ConnectCallback` and `DisconnectCallback` pointer of `hpcd`. The problem is that neither the functions `HAL_PCD_ConnectCallback()` & `HAL_PCD_DisconnectCallback()` themselves, nor those function pointers you mentioned never get called anywhere from code. 

In contrast `HAL_PCD_ResumeCallback()` `hpcd->ResumeCallback` does get executed in `HAL_PCD_IRQHandler()` as seen below:
https://github.com/STMicroelectronics/STM32CubeWB/blob/398d0fbc5ef491da30f777575b911e50e68fa77f/Drivers/STM32WBxx_HAL_Driver/Src/stm32wbxx_hal_pcd.c#L1065-L1069

So my question is still the same - what is the point of these functions then?

Regards

ALABSTM, 2022-04-25:
Hi @fronders,

Actually, both callbacks you pointed out are redefined at application level, as you can see from the code snippet below.

The fact is that, in the case of the `STM32WB` microcontrollers, the **USB** peripheral does **not** support the **V-Bus** sensing. This requires rather using **ADC** or **GPIO**. In both cases, these callbacks could, for instance, be called by the corresponding `ADCx_IRQHandler()` or `EXTIx_IRQHandler()` that should be defined in the [stm32wbxx_it.c](https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Projects/P-NUCLEO-WB55.Nucleo/Applications/USB_Device/CDC_Standalone/Core/Src/stm32wbxx_it.c) file.

I hope this answers your question.

With regards,

https://github.com/STMicroelectronics/STM32CubeWB/blob/398d0fbc5ef491da30f777575b911e50e68fa77f/Projects/P-NUCLEO-WB55.Nucleo/Applications/USB_Device/CDC_Standalone/USB_Device/Target/usbd_conf.c#L355-L388

fronders, 2022-04-25:
Yeah, I saw they are redefined at application level - that's exactly what my initial post mentions. But still these callbacks never get called anywhere.

> The fact is that, in the case of the `STM32WB` microcontrollers, the **USB** peripheral does **not** support the **V-Bus** sensing. This requires rather using **ADC** or **GPIO**. In both cases, [...]

Thanks, that is what I was looking for. Would've been nice if it would have been mentioned somewhere in the sources.

closed 2022-04-25 by fronders