STMicroelectronics/STM32CubeWB #56: Potential variable name clash with C macros

42Bastian, 2022-05-10:
Version: v1.13.3
File: ./Drivers/STM32WBxx_HAL_Driver/Inc/stm32wbxx_hal_pwr_ex.h

```
void              HAL_PWREx_HoldCore(uint32_t CPU);
void              HAL_PWREx_ReleaseCore(uint32_t CPU);
```
Usage of all upper-case for a variable name (may) clash with with macro definitions (which in many style guides shall use all upper-case).

ALABSTM, 2023-10-20:
Hi @42Bastian,

Thank you for this report. In the prototypes you mentioned, `CPU` is rather the argument's name of the function, not a variable. Besides, I can understand from your comment that this actually caused no issue, simply a "potential" risk, which I believe is not one.

Please allow me to close this issue. Thank you for your comprehension and thank you again for this report.

With regards,

closed 2023-10-20 by ALABSTM

42Bastian, 2023-10-20:
It actually is a problem and should be refactored to not use all upper-case letters.

Just the other day a customer got weird compiler errors until we found out that the macro CPU used in our files clashes with the word CPU used in the parameter list.

No matter if "variable" or "parameter" (a parameter is as well a variable in the function body).