STMicroelectronics/STM32CubeWB #57: BLE Event Callbacks not working (ble_events.h)

buchvC, 2022-05-24:
I'm trying to use the ble event callbacks given in "ble_events.c". 

So im implementing the function:

```
aci_gatt_attribute_modified_event( uint16_t Connection_Handle,
                                               uint16_t Attr_Handle,
                                               uint16_t Offset,
                                               uint16_t Attr_Data_Length,
                                               const uint8_t* Attr_Data )
{
  
}

```

but it get never called. The switch case option gets called!

The Description describes that in default all events are masked, so it should not be necessary to mask it with:
`aci_gatt_set_event_mask(0x00000001);`
which changes nothing anyways.

In the manual PM0271 Page 38 is described that the switch case as well as the callback framework can be used for the ble stack events. 
[image]
All Examples use the switch case statement, an issue (https://github.com/STMicroelectronics/STM32CubeWB/issues/24) regarding that is solved, with the comment that one example was added -> I found the example "BLE_Peripheral_Lite_EventCallbacks", but it uses the "switch case" event handler and calls from there the "callbacks" in ble_events.

So to use the callbacks I have to implement the switch case event handler and have to call functions from there, not really how it should be and not really callbacks. In that case the "ble_events" source and header files are useless. So how should it work?

buchvC, 2022-06-21:
I am using the Event Callbacks now with an quite easier way than the ST Example:
(Example only for the LE META Events callback)
```
hci_init(BLE_UserEvtRx, (void*) &Hci_Tl_Init_Conf);

static void BLE_UserEvtRx( void * pPayload )
{
    tHCI_UserEvtRxParam *pParam;
    pParam = (tHCI_UserEvtRxParam *)pPayload;
    void *pckt = &(pParam->pckt->evtserial);
 
    hci_event_pckt *eventPackage;
    eventPackage = (hci_event_pckt*) ((hci_uart_pckt *) pckt)->data;
 
    if (eventPackage->evt == HCI_LE_META_EVT_CODE)
    {
        evt_le_meta_event *metaEvt;
        metaEvt = (evt_le_meta_event*) eventPackage->data;
 
        for(int i = 0; i < HCI_LE_EVENT_TABLE_SIZE; i ++)
        {
            if(hci_le_event_table[i].evt_code == metaEvt->subevent)
            {
                hci_le_event_table[i].process(metaEvt->data);
                return;
            }
        }
    }
}
``` 
Why does ST not use their own implementation for the callbacks in there EventCallback Example?

LebetsVI, 2023-01-14:
Hi  buchvC.
Cool story begined from "https://github.com/STMicroelectronics/STM32CubeWB/issues/24"!
I am amazed at your perseverance and wish you good luck in the fight against STM.
I never use this feature because it has not been implemented normally since its appearance, it is probably still SDK 1.6 or maybe earlier.

RJMSTM, 2023-07-11:
Hello @LebetsVI,

As this issue is a duplicate of https://github.com/STMicroelectronics/STM32CubeWB/issues/24 please allow me thus to close this one and keep the other.

With regards,

RJMSTM, 2023-07-11:
Hello,

Thank you again for having reported this point. It has been fixed with https://github.com/STMicroelectronics/STM32CubeWB/issues/24 in the frame of version 1.12.0 of this firmware.

closed 2023-07-11 by RJMSTM