STMicroelectronics/STM32CubeWB #58: Wrong BLE "full" stack address ?

pdelrot-samea, 2022-05-31:
**Trouble**
BLE_HeartRateFreeRTOS not working on Nucleo board after import and stack programming according the releases notes.

Maybe a wrong value in the documentation.

**Steps to reproduce**
. Import the BLE_HeartRateFreeRTOS project in the IDE. Stack version 1.13.3.
. Compile the Release variant
. Update the FUS of the Nucleo board if needed
. Program the stack "stm32wb5x_BLE_Stack_full_fw". The programmer suggests the address 0x080AC000, whiles the releases note tells 0x080D0000. Choose 0x080D0000.
. Flash the test program compiled before
. Start the software. HRSTM device could not be find by a Android device for example. 

Repeat the same steps, but with the 0x080AC000 address for stack programming.
The HRSTM device created on the Nucleo board could be detected.

I verified twice.

**Describe the set-up**
Laptop, W11 up to date

 * The board (either ST RPN reference or your custom board).
P-NUCLEO-WB55 board

 * IDE or at least the compiler and its version.
STM32CubeIDE 1.9.0 and STM32CubeProgrammer 1.9.0

**Additional context**
Just testing BLE features on new specific boards from our company. I had to get back to the Nucleo board because nothing worked as expected. Even on the Nucleo board the examples didn't worked out of the box.

**Proposed correction**
https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Projects/STM32WB_Copro_Wireless_Binaries/STM32WB5x/Release_Notes.html
should be corrected : 

For STM32WB5xxG(1M) : stm32wb5x_BLE_Stack_full_fw.bin :  0x080AC000

Merci,
Best regards,
Pascal

ASELSTM, 2022-06-27:
Hi @pdelrot-samea,

We have followed the steps you have described to reproduce the issue and unfortunately we were not able to have the same behavior. Actually, the BLE_HeartRateFreeRTOS application is working correctly as described in the readme file and the created HRSTM device was properly detected. 

For a better analysis of you issue would please make sure about the following : 

- Would you please check the version of the FUS you are using. Please make sure you are using the latest version **v1.2.0**.
- Would you please make sure you are using the latest version of the STM32CubeWB firmware **v1.13.3**.

With regards,

ASELSTM, 2023-04-07:
Hi @pdelrot-samea,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-04-07 by ASELSTM