STMicroelectronics/STM32CubeWB #63: STM32WB Middleware WPAN sourcen

ATM-HSW, 2022-10-06:
The readme.txt in the STM32WB Middleware WPAN folder mentions that it is possible to get the clusters library sourcen. How can this be managed?

ATM-HSW, 2022-10-28:
Why is ST not answering? I asked only for an offer from this cube packages. It's mentioned in the cube package itself and it mentioned in the Pdf documentation.

ASELSTM, 2023-07-07:
ST Internal Reference: 156762

RJMSTM, 2023-09-29:
Hello @ATM-HSW,

Refer to Release_Notes Middleware WPAN [Release_Notes ](https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Middlewares/ST/STM32_WPAN/Release_Notes.html)

Depending on the kind of Application targeted, the appropriate Wireless Coprocessor Firmware needs to be loaded.

 **Zigbee** clusters are provided as a library.

- Available on IAR/STM32CubeIDE (stm32wb_zigbee_wb_lib.a) and Keil (stm32wb_zigbee_wb_lib_keil.lib) under Middlewares/ST/STM32_WPAN/zigbee/lib.

- To get access to Zigbee clusters source code, please contact ST local office.

Please allow me to close this issue. You may reopen it at any time if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With Regards,
Rania

closed 2023-09-29 by RJMSTM