STMicroelectronics/STM32CubeWB #65: Firmware Examples project for LL - I2C reader has flaw.

Vince-H, 2022-11-09:
Sample LL code to read from I2C occasionally fails.

Issue is located in lines 442-470 of the following file : 
https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Projects/P-NUCLEO-WB55.Nucleo/Examples_LL/I2C/I2C_TwoBoards_MasterTx_SlaveRx_Init/Src/main.c

It is meant to read I2C data until STOP flag is raised. 
The issue is the ordering of the checks between RXNE and STOP flags.

At the end of a transmission, the **next-to last** byte is read (line 452), then some actions are performed (in particular when checking timeout. It may also get interrupted). It will take some time to go back to the beginning of the loop line 443. 

By the time we get there, the **last** byte may be already received by the HW. In this case we have **both** RXNE and STOP flag raised. 

This will stop the operation and clear the STOP flag without reading the last byte received. 

A proper implementation would check the STOP flag only if/when RXNE is no longer set. 

The issue is real, we've observed it many times. It obviously depends on the CPU, peripheral and bus frequencies. 
It can also be triggered by adding some time-consuming operations between the data read (line 452) and the end of the loop (line 470).

Vince-H, 2022-11-09:
Note : While the issue is reported against STM32CubeWB, the same code is provided with other MCUs. 
Those will most likely have the same issue.
e.g: 
https://github.com/STMicroelectronics/STM32CubeL0/blob/master/Projects/NUCLEO-L073RZ/Examples_LL/I2C/I2C_TwoBoards_MasterTx_SlaveRx/Src/main.c

HBOSTM, 2023-01-11:
Hello @Vince-H 

This example works fine without problems, So, would you please give us more details about how you got this issue?

With Regards,

Vince-H, 2023-01-12:
How did we get the issue ? 
We repeatedly read fixed-size data using the code, and from time to time we observed incomplete reading, missing the last byte.
Again, the like likelihood is low, but can be raised by having frequent interrupts, or longer operation between lines 452 and 470.

HBOSTM, 2023-01-20:
Hello @Vince-H

Would you confirm that you are using the latest version of the STM32CubeWB (**V1.15.0**).

With Regards,

HBOSTM, 2023-05-04:
Hello @Vince-H ,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-05-04 by HBOSTM