STMicroelectronics/STM32CubeWB #67: BLE_Hid Application  - Keyboard with Input and Output Report

rstenet, 2022-12-11:
P-NUCLEO-WB55.Nucleo
STM32CubeIDE: 1.11.0
STM32CubeWB: 1.15.0.
FUS version 1.2.0
Wireless Firmware version 1.15.0 (BLE_Stack_full)

Hi all,

modified the app as follows
1. in **BLE_Hid\core\inc\app_conf.h** changed the debug trace uart to
`#define CFG_DEBUG_TRACE_UART      hw_uart1`

2. in **BLE_Hid\STM32_WPAN\App\hids_app.c** added keyboard report descriptor
```
static uint8_t report_keyboard[] =
{
		0x05, 0x01,	//Usage Page (Generic Desktop),
		0x09, 0x06,	//Usage (Keyboard),
		0xA1, 0x01,	//Collection (Application),

		0x05, 0x07,	//Usage Page (Key Codes);
		0x19, 0xE0,	//Usage Minimum (224),
		0x29, 0xE7,	//Usage Maximum (231),
		0x15, 0x00,	//Logical Minimum (0),
		0x25, 0x01,	//Logical Maximum (1),
		0x75, 0x01,	//Report Size (1),
		0x95, 0x08,	//Report Count (8),
		0x81, 0x02,	//Input (Data, Variable, Absolute), ;Modifier byte
		0x95, 0x01,	//Report Count (1),
		0x75, 0x08,	//Report Size (8),
		0x81, 0x01,	//Input (Constant), ;Reserved byte
		0x95, 0x05,	//Report Count (5),
		0x75, 0x01,	//Report Size (1),

		0x05, 0x08,	//Usage Page (Page# for LEDs),
		0x19, 0x01,	//Usage Minimum (1),
		0x29, 0x05,	//Usage Maximum (5),
		0x91, 0x02,	//Output (Data, Variable, Absolute), ;LED report
		0x95, 0x01,	//Report Count (1),
		0x75, 0x03,	//Report Size (3),
		0x91, 0x01,	//Output (Constant), ;LED report padding
		0x95, 0x06,	//Report Count (6),
		0x75, 0x08,	//Report Size (8),
		0x15, 0x00,	//Logical Minimum (0),
		0x25, 0x65,	//Logical Maximum(101),

		0x05, 0x07,	//Usage Page (Key Codes),
		0x19, 0x00,	//Usage Minimum (0),
		0x29, 0x65,	//Usage Maximum (101),
		0x81, 0x00,	//Input (Data, Array), ;Key arrays (6 bytes)
		0xC0		//End Collection
};
```
changed the **HIDSAPP_Init** accordingly
```
void HIDSAPP_Init(void)
{
  tBleStatus result = BLE_STATUS_INVALID_PARAMS;

  UTIL_SEQ_RegTask( 1<< CFG_TASK_HID_UPDATE_REQ_ID, UTIL_SEQ_RFU, HIDSAPP_Profile_UpdateChar );

  result = HIDS_Update_Char(REPORT_MAP_CHAR_UUID, 
                            0, 
                            0, 
                            sizeof(report_keyboard),
                            (uint8_t *)&report_keyboard);

  if( result == BLE_STATUS_SUCCESS )
  {
    BLE_DBG_APP_MSG("Report Map Successfully Sent\n");
  }
  else 
  {
    BLE_DBG_APP_MSG("Sending of Report Map Failed error 0x%X\n", result);
  }
}
```
modified the **HIDSAPP_Profile_Update** just to send "d" on Button_1 press
```
void HIDSAPP_Profile_UpdateChar(void)
{
  uint8_t hid_buffer[8] = {0,0,7,0,0,0,0,0};
    
  tBleStatus result = BLE_STATUS_INVALID_PARAMS;

  result = HIDS_Update_Char(REPORT_CHAR_UUID,
						  0,
						  0,
						  8,
						  hid_buffer);
  if( result == BLE_STATUS_SUCCESS )
    BLE_DBG_APP_MSG(" d press\n");
  else
	BLE_DBG_APP_MSG(" d press FAILED\n");

  hid_buffer[2] = 0;
  result = HIDS_Update_Char(REPORT_CHAR_UUID,
		  	  	  	  	    0,
							0,
							8,
							hid_buffer);
  if( result == BLE_STATUS_SUCCESS )
	BLE_DBG_APP_MSG(" d release\n");
  else
	BLE_DBG_APP_MSG(" d release FAILED\n");
}
```
3. and changed Human Interface Device Service (HIDS) in **BLE_Hid\STM32_WPAN\App\ble_conf.h**
```
#define BLE_CFG_HIDS_PROTOCOL_MODE_CHAR           0 /**< Support of Protocol Mode Characteristic */
#define BLE_CFG_HIDS_INPUT_REPORT_NB              1 /**< Number of Input Report */
#define BLE_CFG_HIDS_INPUT_WRITE                  0 /**< Support of Input Write property */
#define BLE_CFG_HIDS_OUTPUT_REPORT_NB             1 /**< Number of Output Report */
#define BLE_CFG_HIDS_FEATURE_REPORT_NB            0 /**< Number of Feature Report */
#define BLE_CFG_HIDS_EXTERNAL_REPORT_REFERENCE    0 /**< Support of EXTERNAL REPORT Reference Descriptor */
#define BLE_CFG_HIDS_KEYBOARD_DEVICE              0 /**< Support of BOOT KEYBOARD Report Characteristic */
#define BLE_CFG_HIDS_KEYBOARD_INPUT_WRITE         0 /**< Support of Keyboard Input Write property */
#define BLE_CFG_HIDS_MOUSE_DEVICE                 0 /**< Support of BOOT MOUSE Report Characteristic */
#define BLE_CFG_HIDS_MOUSE_INPUT_WRITE            0 /**< Support of Mouse Input Write property */
#define BLE_CFG_HIDS_REPORT_CHAR                  \
  BLE_CFG_HIDS_INPUT_REPORT_NB   +\
  BLE_CFG_HIDS_OUTPUT_REPORT_NB  +\
  BLE_CFG_HIDS_FEATURE_REPORT_NB
#define BLE_CFG_HIDS_PROTOCOL_MODE                1 /**< Report Protocol Mode */
```

So far so good. Project compiles. Nucleo connects to the PC over BT and on Button_1 press "d" appears.
Now the issue. Upun connection the PC tries to inform the keyboard that the Num Lock is on
[image]
and fails.
When I set **BLE_CFG_HIDS_INPUT_REPORT_NB to 0** the Output Report gets processed correctly
[image]
but as expected "d" is not send to the PC.

How can I get both working?
Thank you.

RJMSTM, 2023-07-20:
Hello,
Hello,

The trouble is localized in the Report Reference Descriptor initialization for each Report Characteristic (especially for for the index in the first byte).
The Report Reference Descriptor is composed by:
- 1 byte for the index of the Report Characteristics with the same type (Input, 
  Output or Feature).
- 1 byte for the report type: 1 for Input, 2 for Output and 3 for Feature.

During HID service and characteristics creation, the index in the Report Reference Descriptors are not correctly initialized.
   
Best regards.

RJMSTM, 2023-07-20:
Hello,

Correction done and available for the next release.

Best regards,
Rania

closed 2023-11-23 by RJMSTM