STMicroelectronics/STM32CubeWB #68: Confusing effects of CFG_HW_RESET_BY_FW flag

hlukasz, 2022-12-21:
`app.h` file from examples defines `CFG_HW_RESET_BY_FW` as follows:

```c
/**
 * When set, this resets some hw resources to set the device in the same state than the power up
 * The FW resets only register that may prevent the FW to run properly
 *
 * This shall be set to 0 in a final product
 *
 */
#define CFG_HW_RESET_BY_FW         1
```

From my understanding it means that when this flag is enabled, initialization code should reset some HW resources when software reset cause is detected. However, when I look at `app_entry.c` where this flag is used, I see:

```c
#if (CFG_HW_RESET_BY_FW == 1)
static void Reset_BackupDomain(void)
{
  if ((LL_RCC_IsActiveFlag_PINRST() != FALSE) && (LL_RCC_IsActiveFlag_SFTRST() == FALSE))
  {
    HAL_PWR_EnableBkUpAccess(); /**< Enable access to the RTC registers */

    /**
     *  Write twice the value to flush the APB-AHB bridge
     *  This bit shall be written in the register before writing the next one
     */
    HAL_PWR_EnableBkUpAccess();

    __HAL_RCC_BACKUPRESET_FORCE();
    __HAL_RCC_BACKUPRESET_RELEASE();
  }

  return;
}
[...]
```

Backup domain reset is executed only when `SFTRST` is NOT SET, so there is other reset source than software reset. This is a little confusing. This behavoiur also leads to a problem when I want to use RTC and Shutdown LPM, because on every wake-up RTC resets.

I guess that there is a bug and `SFTRST` should be checked if is SET. However, if above code is correct, how can I use RTC and Shudown mode if I have to reset backup domain on every non-software reset?

RJMSTM, 2023-07-11:
ST Internal Reference: 157043

ALABSTM, 2023-10-20:
Hi @hlukasz,

Thank you for this report. A fix will be available in a coming release. In the meanwhile, here are some elements from our development teams:
* The purpose from the `CFG_HW_RESET_BY_FW` macro is the debugging, mainly to obtain the same behavior when you press NRST Vs. Power On Reset regarding the RTC behavior.
* As mentioned in the comment `This shall be set to 0 in a final product`, it shall be enabled **only** for debugging **not** for production.

Hence, in your use case where both shutdown mode and RTC are used, you must have the `CFG_HW_RESET_BY_FW` macro set to `0` **not to erase** the RTC configuration at wakeup.

I hope this helps.

With regards,

closed 2023-11-23 by RJMSTM