STMicroelectronics/STM32CubeWB #71: Thread DNS support

agalliazzo, 2023-02-16:
Hi Guys,
I'm working with a P-NUCLEO-WB55 traying to implement an OpenThread network. I'm using the latest stable software package (1.15.0) with STMCubeIDE.

I found some problem browsing response from the `otDnsClientBrowse` function. In particular, in the callback, `otDnsBrowseResponseGetServiceName` return `OT_ERROR_NONE` but no data on the label  (param. 1) is filled.

I tryed to initialize all the buffer to 0 and, after the call, the data is still 0.

I share the code in here to reproduce the problem:
```C
otIp6Address otLocalMLEIDAddress;

void _CP_DNS_BrowseCallback(otError aError, const otDnsBrowseResponse *aResponse, void *aContext){
	#define LBL_LEN	255
	aError = aError;
	otError record_not_found = OT_ERROR_NONE;
	char label[LBL_LEN];
	char svc_name[LBL_LEN];
	uint16_t index = 0;
	otDnsServiceInfo s_info;
	otDnsBrowseResponseGetServiceName(aResponse, svc_name, LBL_LEN);
	if (aError == OT_ERROR_NONE){
		while(otDnsBrowseResponseGetServiceInstance(aResponse, index, label, LBL_LEN) == OT_ERROR_NONE){
			index ++;
			if(otDnsBrowseResponseGetServiceInfo(aResponse, label, &s_info) == OT_ERROR_NONE){
				if(strstr(label, "CoAP_MQTT_Bridge") != 0){
					memcpy(&otMainServerAddress, &s_info.mHostAddress, sizeof(otIp6Address));
					break;
				}
			}
		}
	}
}

void CP_InitializeSRP_DNS(coap_protocol_info_t * ci) {
	otError error = OT_ERROR_NONE;
	otDnsTxtEntry txt_entry;

	_CP_GetOMR();

	// At this point, thanks to the call to _CP_GetOMR(),
	// otLocalMLEIDAddress should contain the OMR for the thread node

	// Registering the local service to the SRP server

	service.mInstanceName = ci->EUI64; // Let's start with the EUI64 as service name
	service.mName = "_coap._udp"; // I'm publishing a CoAP service
	service.mPort = OT_DEFAULT_COAP_PORT;
	service.mPriority = 1;
	service.mWeight = 1;
	service.mSubTypeLabels = NULL;
	service.mTxtEntries = &txt_entry; // This is a test I did, I need futher investigation about the utility in my case
	service.mNumTxtEntries = 1;

	otServiceConfig conf;
	otNetworkDataIterator it = 0;
	while (conf.mServiceData[0] != 0x5d) {
		error = otNetDataGetNextService(NULL, &it, &conf);
	}
	memcpy(&SRP.mAddress.mFields.m8, &conf.mServerConfig.mServerData, 16);
	SRP.mPort = conf.mServerConfig.mServerData[16] << 8 | conf.mServerConfig.mServerData[17];

	error = otSrpClientSetHostAddresses(NULL, &otLocalMLEIDAddress, 1);
	error = otSrpClientAddService(NULL, &service);
	error = otSrpClientSetHostName(NULL, ci->EUI64);
	error = otSrpClientStart(NULL, &SRP);

       // At this time the service is properly registered on the SRP server, we need to know if a CoAP to MQTT bridge is available in the network

	// Querying the server for a CoAP to MQTT bridge

	otDnsQueryConfig * dns_cfg = otDnsClientGetDefaultConfig(NULL); 
	dns_cfg->mMaxTxAttempts = 5;
	dns_cfg->mResponseTimeout = 1000;
	memcpy(&dns_cfg->mServerSockAddr, &SRP, sizeof(otSockAddr));
	dns_cfg->mServerSockAddr.mPort = 53;

	otDnsClientBrowse(NULL, "_coap._udp.default.service.arpa", _CP_DNS_BrowseCallback, ci, dns_cfg);
}
```

I tesded the CLI example you provide with the software pack and that works fine, I suppose the problem somewhere between the M4 and M0 core but, of course, I can be wrong!

If I'm doing something wrong please feel free to comment! I'm quite new to both OpenThread and STM32 microcontroller.

Thanks in advance
Alessio

agalliazzo, 2023-02-16:
FYI: also `otDnsAddressResponseGetAddress` on the callback from `otDnsClientResolveAddress` seems broke... Maybe it's still me that I'm missing something...

ASELSTM, 2023-07-17:
ST Internal Reference: 157403

ALABSTM, 2023-10-20:
Hi @agalliazzo,

I hope you are fine. Please find below the answer from our development teams:

### Regarding the otDnsClientGetDefaultConfig() method

* In the code snippet given, the `otDnsClientGetDefaultConfig()` method is **misused**. Indeed, in the code snippet:

```c
otDnsQueryConfig * dns_cfg = otDnsClientGetDefaultConfig(NULL);
dns_cfg->mMaxTxAttempts = 5;
dns_cfg->mResponseTimeout = 1000;
memcpy(&dns_cfg->mServerSockAddr, &SRP, sizeof(otSockAddr));
dns_cfg->mServerSockAddr.mPort = 53;
```

  * But `otDnsClientGetDefaultConfig()` returns a **constant** pointer that should **not** be overwritten. It is better to write:

```c
const otDnsQueryConfig * dns_cfg = otDnsClientGetDefaultConfig(NULL); 
otDnsQueryConfig defaultConfigSet;

memcpy(&defaultConfigSet, dns_cfg, sizeof(otDnsQueryConfig));
        
defaultConfigSet.mMaxTxAttempts = 5;
defaultConfigSet.mResponseTimeout = 1000;
memcpy(&defaultConfigSet.mServerSockAddr, &SRP, sizeof(otSockAddr));
defaultConfigSet.mServerSockAddr.mPort = 53;

OtDnsClientSetDefaultConfig(NULL, &defaultConfigSet);
```

### Regarding the otDnsBrowseResponseGetServiceName() method

* The prototype of the `otDnsBrowseResponseGetServiceName()` method is:

```c
otError otDnsBrowseResponseGetServiceName(const otDnsBrowseResponse *aResponse, char *aNameBuffer, uint16_t aNameBufferSize)
```

* From our side, the tests showed that:
  * the `aNameBuffer` parameter is not empty,
  * the `otIp6Address` retruned by the `otDnsAddressResponseGetAddress()` is a valid one.

* Would you mind changing your DNS query configuration and tell us if it solves the problem?

I hope this helps.

With regards,

closed 2023-11-23 by RJMSTM