STMicroelectronics/STM32CubeWB #72: Feature Request: Allow Device Information Service generation without requiring another ST defined BLE service

Bradley-Cooley, 2023-03-11:
Currently it seems the device information service implementation provided by ST through CubeMX code generation is hidden behind usage of other services. I am not creating a, for example, Heart Rate device, but I do want to use the DIS service. I unfortunately cannot use STs DIS service through CubeMX because there is no option to generate it outside the context of other services, and the autogeneration step will delete the DIS files if I remove the i.e. Heart Rate service.

RJMSTM, 2023-08-15:
ST Internal Reference: 159311

RJMSTM, 2023-09-29:
Hello @Bradley-Cooley,

If I well understand the User try to create a WB55 BLE application that advertise one service the Device Information by deleting in the WB55 BLE Heart Rate application the Heart Rate service.

The best way to create a WB55 BLE application with CubeMx is to create the application from scratch.
There is a wiki page that can help to create a custom WB55 BLE application from scratch with CubeMx at the following address:
https://wiki.st.com/stm32mcu/wiki/Connectivity:STM32WB_BLE_STM32CubeMX

Let me know if with this wiki page you can create his custom WB55 BLE application.

closed 2023-09-29 by RJMSTM