STMicroelectronics/STM32CubeWB #73: Wrong Start Address for stm32wb1x_BLE_Stack_light_fw

TillerBelotti, 2023-03-12:
Setup: NUCLEO-WB15CC rev 2 (GU2). STM32CubeProgrammer GUI v2.13.0

Problem: According to v1.16.0 release notes, start address for stm32wb1x_BLE_Stack_light_fw is 0x08029000. I think it is wrong.

1. FUS does not accept stm32wb1x_BLE_Stack_light_fw install in 0x08029000.

2. As in v1.15.0 the address was 0x08028000 and stm32wb1x_BLE_Stack_light_fw.bin from both versions have the same size, I installed it using old address (0x08028000) and it worked just fine.

3. I've tried to install it using the new address (0x08029000) many times in different boards, using bootloader and STM32CubeProgrammer CLI and the error was always present.

RJMSTM, 2023-06-08:
ST Internal Reference: 154415

RJMSTM, 2023-07-11:
Hello,

Yes, the binary stm32wb1x_BLE_Stack_light_fw.bin  is correct however the recommended flash address has not been updated in RN(v1.16.0), it should have been corrected as:
stm32wb1x_BLE_Stack_light_fw.bin: 0x08029000 -->  0x08028000 due to binary size increase with padding before encryption. 
All other addresses remain unchanged.

This issue has been fixed in the frame of version v1.17.0 of the STM32CubeWB. Please allow me then to close this issue.

With regards,
Rania

closed 2023-07-11 by RJMSTM