STMicroelectronics/STM32CubeWB #74: SHCI_C2_FLASH_StoreData not supported feature

agalliazzo, 2023-03-14:
Hi guys,
Using the latest STM32CubeWB pacakge (1.16.0) it seems that is not possible to store non volatile information to flash. I never tested on previous releases so I don't really know if it is a new issue or a old problem.

Executing
```C
SHCI_CmdStatus_t status = SHCI_C2_FLASH_StoreData(THREAD_IP);
```
result in SHCI_ERR_UNSUPPORTED_FEATURE and it's quite strange cause, by openthread specification, thread need some non volatile data to be stored. Also your AN5289 states that is up to the user application to call SHCI_C2_FLASH_StoreData on, for example, commissioning. 

This problem makes the STM32WB55 as a client device with commissioner cause there seems no way to store network information at the reboot. This also introduces other problems like SRP not properly working due to loosing DNS key to update the IP address to the SRP server.

agalliazzo, 2023-08-05:
Hi guys, do you have any update on this issue? A bugfix release date will be really helpful to plan my developing efforts!

RJMSTM, 2023-09-29:
Hello @agalliazzo,

Our Thread framework supports 'non-volatile data' storage.
The best way to proceed is to run the following application example named Thread_NVM (Under P-NUCLEO-WB55.Nucleo\Applications\Thread) which shows how to save persistent.

Regards,
Rania

closed 2023-09-29 by RJMSTM