STMicroelectronics/STM32CubeWB #77: ADC doesn't work when CFG_LPM_SUPPORTED is set to 1.

Hubert97, 2023-05-08:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * STM32WB15
 * STM32CubeIDE:v 1.11.2.
 * 
**Additional context**
I've added simple adc configuration iside "HRSAPP_Measurement" function.
"
		HAL_ADC_MspInit(&hadc1);
		HAL_ADC_Start(&hadc1);
		HAL_ADC_PollForConversion(&hadc1, 10);
		int measurement = HAL_ADC_GetValue(&hadc1);
		HAL_ADC_Stop(&hadc1);
		HAL_ADC_MspDeInit(&hadc1);
"
And ADC always retunrs 0.
When CFG_LPM_SUPPORTED is set to 0 it works properly.

Hubert97, 2023-05-14:
I'd like to mention that the same code invoked inside main function (pre initialisation of middleware) works correctly.

RJMSTM, 2023-05-15:
Hello @Hubert97,

Thank you for your contribution. Would it be possible to share the project you have used to reproduce the issue in order to allow a better analysis of the problem.

Regards,

Hubert97, 2023-05-15:
Sure, how would you like me to share it?

Hubert97, 2023-05-16:
I've packed the whole directory. To open simply replace directory inside this repository.

[BLE_HeartRate.zip](https://github.com/STMicroelectronics/STM32CubeWB/files/11485514/BLE_HeartRate.zip)

Hubert97, 2023-05-16:
Hello, [RJMSTM](https://github.com/RJMSTM)
I've packed the whole directory. To open simply replace directory inside this repository.
[BLE_HeartRate.zip](https://github.com/STMicroelectronics/STM32CubeWB/files/11485806/BLE_HeartRate.zip)
Regards,
Hubert

Hubert97, 2023-05-22:
Hello, [RJMSTM](https://github.com/RJMSTM)
Are there any updates?
Regards,
Hubert

RJMSTM, 2023-05-22:
Hello @Hubert97,

First, sorry for my tardy reply.
Second, when I upload your project, I have 58 errors can you please give me the path when I will replace it to reproduce the problem.

Regards,
Rania

Hubert97, 2023-05-22:
I have it inside: STM32CubeWB-master\Projects\NUCLEO-WB15CC\Applications\BLE\BLE_HeartRate

Hubert97, 2023-05-22:
Hello, [RJMSTM](https://github.com/RJMSTM)
I have it inside: STM32CubeWB-master\Projects\NUCLEO-WB15CC\Applications\BLE\BLE_HeartRate
When i import the project import path is: `STM32CubeWB-master\Projects\NUCLEO-WB15CC\Applications\BLE\BLE_HeartRate\STM32CubeIDE`
Regards,
Hubert

RJMSTM, 2023-05-22:
Hello @Hubert97,

Which firmware version have you used? Because I still get errors when I build the project.

Regards,
Rania

Hubert97, 2023-05-22:
Hello, [RJMSTM](https://github.com/RJMSTM)
I've pushed my version to repository https://github.com/Hubert97/STM32CubeWB. I'm using 
```
 STM32CubeIDE

Version: 1.12.1
Build: 16088_20230420_1057 (UTC)
```
It builds fine on my end.  Path is as prevoiusly `STM32CubeWB-master\Projects\NUCLEO-WB15CC\Applications\BLE\BLE_HeartRate`
Regards,
Hubert

RJMSTM, 2023-05-22:
Hello @Hubert97 

And how did you find that ADC always returns 0 in this case?  

Regards,
Rania

Hubert97, 2023-05-22:
Hello, [RJMSTM](https://github.com/RJMSTM)
I made a custom BLE app. However you can also use ST BLE sensor app on Google play. If you create offset to what adc returns ( ie add 1 to measurement) it will display only the offset. 
Regards,
Hubert

Hubert97, 2023-05-24:
Hello, [RJMSTM](https://github.com/RJMSTM)
DId you manage to run the aplication?
After some investigation i think the problem lies in the way the cpu state is saved than resumed. for standby reasons. 
```

/* These 2 functions are designed to save and then restore CPU context. */
  .global CPUcontextSave
  .type CPUcontextSave, %function
CPUcontextSave:
  PUSH   { R4 - R12, LR }        /* store R4 to R12 and LR (10 words) onto the stack */
  LDR    R4, =STBY_BackupMSP     /* load address of STBY_BackupMSP into R4 */
  MOV    R3, SP                  /* load the stack pointer into R3 */
  STR    R3, [R4]                /* store the MSP into STBY_BackupMSP */
  DSB
  WFI                            /* all saved, trigger deep sleep */

CPUcontextRestore:
  /* Even if we fall through the WFI instruction, we will immediately
   * execute a context restore and end up where we left off with no
   * ill effects.  Normally at this point the core will either be
   * powered off or reset (depending on the deep sleep level). */
  LDR    R4, =STBY_BackupMSP     /* load address of STBY_BackupMSP into R4 */
  LDR    R4, [R4]                /* load the SP from STBY_BackupMSP */
  MOV    SP, R4                  /* restore the SP from R4 */
  POP    { R4 - R12, PC }        /* load R4 to R12 and PC (10 words) from the stack */
/* end of specific code section for standby */
```
inside  startup_stm32wb15ccux.h
It looks fine but why not all registers? wouldn't Pushing from R0 to R12 make mores sense?
Also i can't figure why suspending cpu and resuming its operation interfere with adc initialisation.  It's not like we are resuming its operation. 
Regards,
Hubert

Hubert97, 2023-05-30:
Hello, [RJMSTM](https://github.com/RJMSTM)
Any progress on the issue? I see that you have a lot of other things going on. I've already tried a lot of things and i seam to have run out of ideas. If you could give me some pointers where to look at i might be able to figure it out. I'd like to help with solving this but i lack understanding of the whole system.
Regards,
Hubert

RJMSTM, 2023-05-30:
Hello @Hubert97,

The ADC looks well configured, but I don't understand when **HRSAPP_Measurement** is called. I don't know what it's supposed to do. I think I have ask someone who knows this example well.

Regards,
Rania

Hubert97, 2023-05-30:
Hello, [RJMSTM](https://github.com/RJMSTM)
Thank you for your reply. Please do. I'd be grateful. 
Regards,
Hubert

RJMSTM, 2023-05-31:
ST Internal Reference: 153651

RJMSTM, 2023-06-05:
Hello @Hubert97,

Thank you for this contribution, this point has been reported to our development teams. I will get back to you as soon as I have any updates.

With regards,
Rania

Hubert97, 2023-06-20:
Hello, [RJMSTM](https://github.com/RJMSTM)
I'm aware how this task works. What is your point? I've modified the code so measurements would be real, from ADC. And my issue is that it only works when CFG_LPM_SUPPORTED is set to 0. As i initialised ADC from scratch i can't understand why this wouldn't work even if everything would be reset by going to sleep.
Regards,
Hubert

RJMSTM, 2023-07-11:
Hello @Hubert97 

Would it be possible to share the project you have used?

Thanks

Hubert97, 2023-07-12:
Hello, [RJMSTM](https://github.com/RJMSTM)
I've shared it in here.
https://github.com/Hubert97/STM32CubeWB
Regards,
Hubert

> Hello, [RJMSTM](https://github.com/RJMSTM) I've pushed my version to repository https://github.com/Hubert97/STM32CubeWB. I'm using
>
> ```
>  STM32CubeIDE
>
> Version: 1.12.1
> Build: 16088_20230420_1057 [...]

Hubert97, 2023-07-21:
Hello, [RJMSTM](https://github.com/RJMSTM)
Any updates on the issue? Is there anything i could help you with?
Regards,
Hubert

RJMSTM, 2023-07-21:
Hello @Hubert97,
Have you checked ADC clock configuration when output from Stop mode ?
Do you encounter same issue when forcing Sleep mode only ?
Please can you attach a ZIP of project to BZ ticket,

RJMSTM, 2023-09-29:
Hello @Hubert97,

Please allow me to close this issue if you are convinced. You may reopen it at any time if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With Regards,
Rania

closed 2023-09-29 by RJMSTM

Hubert97, 2023-10-29:
Hello, [RJMSTM](https://github.com/RJMSTM)
I've experimented with clock configuration on my own. And i've noticed that even when i put whole configuration as is from main (SystemClock_Config, PeriphCommonClock_Config,  MX_GPIO_Init and MX_ADC1_Init). In my mind it should restore configuration to state before going into stop mode thus enabling adc to do measurement. Should that be the case? Or am i understanding something wrongly? 
Regards,
Hubert