STMicroelectronics/STM32CubeWB #78: Issue with SRP client

edward62740, 2023-05-11:
**Describe the set-up**
STM32WB5MMG (Custom board)
STM32CubeWB FW 1.16.0 (stm32wb5x_Thread_MTD_fw.bin 1.16.0, stm32wb5x_FUS_fw.bin 1.2.0)

**Describe the bug**
This issue concerns the behavior of the SRP client. Once the device registers a service with the SRP server, it will eventually fail to re-register/update once the SRP lease expires or some other update to the SRP record is needed.

The ip6 address pointer in `otDnsAddressResponseGetAddress` and some other stuff in DNS-related callbacks appear to not get updated by the M0. It is hard to determine the cause as the operation of M0 is largely opaque. This issue appears to be related to #71 and #74, perhaps due to some issue with NVM.

Setting the macro OPENTHREAD_SETTINGS_RAM in the `openthread_api_config_mtd.h` appears to be a sufficient workaround, and resolves this particular issue for me.

**How To Reproduce**
This behavior can be triggered by restarting the SRP server.

RJMSTM, 2023-07-27:
ST Internal Reference: 158188

RJMSTM, 2023-11-28:
Hello,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.18.0 of the STM32CubeWB published recently on GitHub. Thank you again for having reported.

closed 2023-11-28 by RJMSTM

agalliazzo, 2024-01-23:
@RJMSTM
Sorry to bother you again on this point but it is still not clear:

> Regarding SRP issue when restarting from persistence, it was found that SRP record key is not included in Thread Active Dataset which was saved calling otDatasetGetActiveTlvs.
> SRP record key is hidden within Thread persistent data buffer.
> Backup of full Thread persistent data buffer was added to Matter SW.
> Restart from persistence is now ok whatever SW/HW reset

How "Backup of full Thread persistent data buffer" is achieved? Trying to understand your words I suppose I have to call some specific function but where? I try to navigate in the commit changes and I did not see anything...

Best
Alessio

ALABSTM, 2024-05-27:
Hi @agalliazzo,

Would you mind reposting your question in a new issue please? This makes it easier to track and process.

Thank you,