STMicroelectronics/STM32CubeWB #80: HAL_UARTEx_ReceiveToIdle_DMA(). DMA in ring mode. When the HAL_UARTEx_RxEventCallback() interrupt is called, the HAL_UARTEx_GetRxEventType() function does not return the HAL_UART_RXEVENT_IDLE status if the index of the last received byte is equal to the size of the HAL_UARTEx_ReceiveToIdle_DMA.

hayar2, 2023-06-30:
**Describe the set-up**
 * The board - [STM32WB55CC](https://www.st.com/en/microcontrollers-microprocessors/stm32wb55cc.html)
 * IDE - STM32CubeIDE version: 1.12.0. 
 * HAL version - 1.16.0

**Describe the bug**
I use the HAL_UARTEx_ReceiveToIdle_DMA() function. DMA in ring mode. 
When the HAL_UARTEx_RxEventCallback() interrupt is called, the HAL_UARTEx_GetRxEventType() function does not return the  HAL_UART_RXEVENT_IDLE status if the index of the last received byte is equal to the size of the HAL_UARTEx_ReceiveToIdle_DMA.

**How To Reproduce**

<details> 
<summary>.io configuration:
</summary>

```
#MicroXplorer Configuration settings - do not modify
CAD.formats=
CAD.pinconfig=
CAD.provider=
Dma.LPUART1_RX.0.Direction=DMA_PERIPH_TO_MEMORY
Dma.LPUART1_RX.0.EventEnable=DISABLE
Dma.LPUART1_RX.0.Instance=DMA1_Channel3
Dma.LPUART1_RX.0.MemDataAlignment=DMA_MDATAALIGN_BYTE
Dma.LPUART1_RX.0.MemInc=DMA_MINC_ENABLE
Dma.LPUART1_RX.0.Mode=DMA_NORMAL
Dma.LPUART1_RX.0.PeriphDataAlignment=DMA_PDATAALIGN_BYTE
Dma.LPUART1_RX.0.PeriphInc=DMA_PINC_DISABLE
Dma.LPUART1_RX.0.Polarity=HAL_DMAMUX_REQ_GEN_RISING
Dma.LPUART1_RX.0.Priority=DMA_PRIORITY_LOW
Dma.LPUART1_RX.0.RequestNumber=1
Dma.LPUART1_RX.0.RequestParameters=Instance,Direction,PeriphInc,MemInc,PeriphDataAlignment,MemDataAlignment,Mode,Priority,SignalID,Polarity,RequestNumber,SyncSignalID,SyncPolarity,SyncEnable,EventEnable,SyncRequestNumber
Dma.LPUART1_RX.0.SignalID=NONE
Dma.LPUART1_RX.0.SyncEnable=DISABLE
Dma.LPUART1_RX.0.SyncPolarity=HAL_DMAMUX_SYNC_NO_EVENT
Dma.LPUART1_RX.0.SyncRequestNumber=1
Dma.LPUART1_RX.0.SyncSignalID=NONE
Dma.LPUART1_TX.1.Direction=DMA_MEMORY_TO_PERIPH
Dma.LPUART1_TX.1.EventEnable=DISABLE
Dma.LPUART1_TX.1.Instance=DMA1_Channel4
Dma.LPUART1_TX.1.MemDataAlignment=DMA_MDATAALIGN_BYTE
Dma.LPUART1_TX.1.MemInc=DMA_MINC_ENABLE
Dma.LPUART1_TX.1.Mode=DMA_NORMAL
Dma.LPUART1_TX.1.PeriphDataAlignment=DMA_PDATAALIGN_BYTE
Dma.LPUART1_TX.1.PeriphInc=DMA_PINC_DISABLE
Dma.LPUART1_TX.1.Polarity=HAL_DMAMUX_REQ_GEN_RISING
Dma.LPUART1_TX.1.Priority=DMA_PRIORITY_LOW
Dma.LPUART1_TX.1.RequestNumber=1
Dma.LPUART1_TX.1.RequestParameters=Instance,Direction,PeriphInc,MemInc,PeriphDataAlignment,MemDataAlignment,Mode,Priority,SignalID,Polarity,RequestNumber,SyncSignalID,SyncPolarity,SyncEnable,EventEnable,SyncRequestNumber
Dma.LPUART1_TX.1.SignalID=NONE
Dma.LPUART1_TX.1.SyncEnable=DISABLE
Dma.LPUART1_TX.1.SyncPolarity=HAL_DMAMUX_SYNC_NO_EVENT
Dma.LPUART1_TX.1.SyncRequestNumber=1
Dma.LPUART1_TX.1.SyncSignalID=NONE
Dma.Request0=LPUART1_RX
Dma.Request1=LPUART1_TX
Dma.Request2=USART1_RX
Dma.Request3=USART1_TX
Dma.RequestsNb=4
Dma.USART1_RX.2.Direction=DMA_PERIPH_TO_MEMORY
Dma.USART1_RX.2.EventEnable=DISABLE
Dma.USART1_RX.2.Instance=DMA1_Channel1
Dma.USART1_RX.2.MemDataAlignment=DMA_MDATAALIGN_BYTE
Dma.USART1_RX.2.MemInc=DMA_MINC_ENABLE
Dma.USART1_RX.2.Mode=DMA_CIRCULAR
Dma.USART1_RX.2.PeriphDataAlignment=DMA_PDATAALIGN_BYTE
Dma.USART1_RX.2.PeriphInc=DMA_PINC_DISABLE
Dma.USART1_RX.2.Polarity=HAL_DMAMUX_REQ_GEN_RISING
Dma.USART1_RX.2.Priority=DMA_PRIORITY_LOW
Dma.USART1_RX.2.RequestNumber=1
[... 274 lines cut ...]
There's a mistake. The HAL_UARTEx_RxEventCallback function is called 1 time.  With the status HAL_UART_RXEVENT_TC.

**Additional context**
perhaps this is not a bug, but a feature. But in the current code it is impossible to determine the end of reception if HAL_UART_RXEVENT_IDLE == HAL_UART_RXEVENT_TC.

the bug occurs around in [this place](https://github.com/STMicroelectronics/STM32CubeWB/blob/d7e2d6d82b45b0323367351daa7270568f981a0f/Drivers/STM32WBxx_HAL_Driver/Src/stm32wbxx_hal_uart.c#L2420).  If you add to the condition 
```c
(nb_remaining_rx_data <= huart->RxXferSize)
```
then everything will work in cyclic DMA mode. But the normal DMA mode will break.

**Screenshots**
<details>
<summary>  half complete: 
</summary>

[image]
</details>

<details> 
<summary>  full complete:
</summary>

[image]
<details>

RJMSTM, 2023-07-04:
Hello @hayar2,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

ASELSTM, 2023-08-16:
Hi @hayar2,

If I understand well, your point is that by the end of the transfer `HAL_UARTEx_GetRxEventType()` does not return `HAL_UART_RXEVENT_IDLE` and it returns only `HAL_UART_RXEVENT_TC`.

The `HAL_UART_RXEVENT_IDLE` should be returned when an Idle event occurs before the reception has been completed as mentioned in the function description 

https://github.com/STMicroelectronics/STM32CubeWB/blob/d7e2d6d82b45b0323367351daa7270568f981a0f/Drivers/STM32WBxx_HAL_Driver/Src/stm32wbxx_hal_uart_ex.c#L965-L966

And as per the description of the `HAL_UARTEx_ReceiveToIdle_DMA()` function,  UART will receive an amount of data in DMA mode till either the expected number of data is received or an IDLE line event occurs.

https://github.com/STMicroelectronics/STM32CubeWB/blob/d7e2d6d82b45b0323367351daa7270568f981a0f/Drivers/STM32WBxx_HAL_Driver/Src/stm32wbxx_hal_uart_ex.c#L885-L889

Actually, the IDLE line event is triggered if UART RX line is inactive for one frame. The frame time depends on your baudrate. Higher baudrate, lower frame.

In your case, the IDLE line event has been detected since all expected data are not received yet. Indeed, if at least one data has already been received, IDLE event is to be notified to user for.

Using the same code, when transferring a data size equal to 8, as the number of received data is exactly equal to the amount of data expected to be received, no idle line event will be generated.  The `HAL_UARTEx_GetRxEventType()` function will return only half transfer and transfer complete events.  The `HAL_UART_RXEVENT_HT` is generated when the fourth character is received and `HAL_UART_RXEVENT_TC` is generated by the end of transfer.

```text
Size = 4, RxEventType = HT
Size = 8, RxEventType = TC
```

I hope I've been able to answer your question. 

With regards,

hayar2, 2023-08-17:
okay, how to detect the end of a package in a cyclic mode when HAL_UARTEx_ReceiveToIdle_DMA Size equal to 8 bytes. if received 16 bytes, 24 bytes, 32 bytes?

sorry for my English. If something is unclear, please tell me about it.

ASELSTM, 2023-08-17:
Hi @hayar2,

The `HAL_UART_RXEVENT_TC` event could notify you the end of transfer it is trigged once the reception has been completed and the expected number of data has been received.

With regards,

hayar2, 2023-08-23:
Changed the HAL_UARTEx_RxEventCallback() function. Added the output of the received bytes to another uart(printf):
<details>

``` c
void HAL_UARTEx_RxEventCallback(UART_HandleTypeDef *huart, uint16_t Size)
{
  debug_pin_toggle(1);

  printf("Size= %i, RxEventType = ", Size);

  HAL_UART_RxEventTypeTypeDef rxEventType;
  rxEventType = HAL_UARTEx_GetRxEventType(huart);
  switch (rxEventType)
  {
    case HAL_UART_RXEVENT_IDLE:
      printf( "IDLE\n\r" );
      break;

    case HAL_UART_RXEVENT_HT:
      printf( "HT\n\r" );
      break;

    case HAL_UART_RXEVENT_TC:
      printf( "TC\n\r" );
      break;

    default:
      printf( "???\n\r" );
      break;
  }

  static int startIdx = 0;
  int endIdx = Size;

//  if ( startIdx == endIdx )
//    return;

  if ( endIdx < startIdx )
  {
    startIdx = 0;
  }

  for (int idx = startIdx; idx < endIdx; idx++)
  {
    printf("0x%X ", huart->pRxBuffPtr[idx]);
  }
  printf("\n\r\n\r");

  startIdx = endIdx;
}
```
</details>

when I send 24 bytes, the following happens:
<details >
<summary>image</summary>

[image]

</details>

It turns out I received 3 packages? But in fact I only accepted 1 package.

ASELSTM, 2023-09-25:
Hi @hayar2,

I didn't get your question precisely. Would you please give me more details ? 

With regards,

hayar2, 2023-10-11:
of course. how to detect the end of a package in this case?
[image]

KRASTM, 2024-04-08:
ST Internal Reference: 178447

RJMSTM, 2024-05-13:
Fixed in : 273007f20a5c7bca2698668905a71515cf2116fd

closed 2024-05-13 by RJMSTM