STMicroelectronics/STM32CubeWB #85: Custom_STM_App_Update_Char() could include a size parameter for supporting characteristics with variable length

marcelobarrosufu, 2023-12-26:
In CubeIDE we may define characteristics with variable length (Length characteristic as variable and value length with maximum size). 

However, when the code is generated (Custom P2P Server disabled and Custom Template is enabled in BLE Applications and services) [the generated code ](https://github.com/STMicroelectronics/STM32CubeWB/blob/7564de628d7e4ec78bd5442d288697a8604d6f31/Middlewares/ST/STM32_WPAN/ble/svc/Src/template_stm.c#L260C19-L260C19) for updating characteristic values do not have any length parameter. 

The default size is controlled by local variables inside custom_stm.c and we do have access to the CustomContext as well since it is static inside custom_stm.c. So, we can´t change the size from external code.

The default implementation for Custom_STM_App_Update_Char is below:

`
tBleStatus Custom_STM_App_Update_Char(Custom_STM_Char_Opcode_t CharOpcode, uint8_t *pPayload)
{
 // ...
}
`

This way, we need to reimplement Custom_STM_App_Update_Char inside custom_stm.c, probably using some protection area. How about to add a support for size in this function call ? Something like:

`
tBleStatus Custom_STM_App_Update_Char(Custom_STM_Char_Opcode_t CharOpcode, uint8_t *pPayload, uint16 size)
{
 // ...
}
`

Thanks a lot !

[image]

[image]

RJMSTM, 2023-12-29:
ST Internal Reference: 169675

RJMSTM, 2024-07-02:
Hello,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.20.0 of the STM32CubeWB published recently on GitHub. Thank you again for having reported.

closed 2024-07-02 by RJMSTM