STMicroelectronics/STM32CubeWB #86: SRP still not working for me

agalliazzo, 2024-01-30:
**Describe the set-up**
 * I'm working on a NUCLEO-WB55RG
 * STM32CubeIde updated at the last version available
 * STM32Cube_FW_WB_V1.18.0

** Actual network configuration **
 * OT Border router at the last commit, a thread network with 36 devices, 17 routers, 1 leader and other are childs.
 * Some childs are based on STM32WB55 (custom board), other devices (all the routers) on the NRF52840
 * 

**Describe the bug**
SRP client still not working also if #78  is marked as resolved. In particular, I can register the first time, then I try to save NVM data (and it works, I check if saved data is in the FLASH using STMCubeProgrammer and the data match) but at the next power-on SRP fail to register due to loosing the key as suggested in [this](https://github.com/openthread/openthread/issues/8056) old thread

**How To Reproduce**
1. Started a new project from scratch from Thread_SED_Coap_FreeRTOS example
2. Added features like APP_NVM_Init, Save, Write and Read from the Thread_NVM as suggested in #74 
3. Added function to get the GUA address
4. Added a function to get the SRP server socket (address + port)
5. Added a function to register the client to the server
6. Modify the var OT_NvmRamBuffer to be placed at RAM2 as described on line 44 of Thread_SED_Coap_FreeRTOS\Core\Inc\hw_conf.h  (tested also with the original configuration present in the Thread_NVM sample)
 
**Steps in code**
1. Initialize the connection choosing between hard-coded configuration or NVM data based on the presence of the panid value (if 0xffff memory is empty)
2. Wait to state notification change to Child or Router (it's an MTD, so it will be a child)
3. Once device is a child wait 10s to ensure everything is properly setted up (this is only a safety delay, networkdata should already be available, but waiting is not a problem)
4. Get the device GUA
5. Get the SRP server socket
6. Register a service with the GUA in the SRP server socket.
9. After about 1s from the first sucesfull SRP registration, with a FreeRTOS timer, a function call NVM_Save(). At this point data is stored starting from address 0x8000000 + 500K (I edited the linker script to reserve this flash area for NVM storage, this was missing on the Thread_NVM example)
10. Power off the board and wait 5s
11. Power on the board again, SRP registration is refused by the server

My code:
OHCS_Protocol.h
```C 
#ifndef OHCS_PROTOCOL_H
#define OHCS_PROTOCOL_H

#include "FreeRTOS.h"
#include "timers.h"
#include "srp_client.h"
#include "openthread_api_wb.h"

#define TMR_ON_CONNECT_ID	1

#define OT_IN NULL
#define get_default_ot_isntance() NULL

#define CONTAINER_OF(ptr, type, field) ((type *)(((char *)(ptr)) - offsetof(type, field)))

typedef union eui64{
    uint8_t bytes[8];
    uint32_t int16[4];
    uint32_t int32[2];
} eui64_t;

typedef struct {
	void (*cb_SaveNVMData)();
	TimerHandle_t state_change_timer;
	TimerHandle_t save_data_timer;
	otIp6Address OMR;
	otSockAddr SRP_server_address;
	otSrpClientService service;
	eui64_t eui64;

	char * hostname;

} oh_coap_server_context_t;

void OHCS_Initialize(oh_coap_server_context_t * ctx);
[... 1422 lines cut ...]
		APP_DBG("APP_THREAD_NVM_Write FAIL!");
	}
	APP_DBG("--- APP_THREAD_NvmSave END ---");
}
/* USER CODE END FD_WRAP_FUNCTIONS */
```

app_entry.c - Relevant part only
``` C
static void APPE_SysEvtReadyProcessing(void)
{
  /* Traces channel initialization */
  TL_TRACES_Init();
  /* Configuration to CPU2 */
  SHCI_C2_CONFIG_Cmd_Param_t config_param = {0};
  /* Set the address that will be used by OT stack for NVM data management */
  config_param.ThreadNvmRamAddress = (uint32_t)OT_NvmRamBuffer;
  (void)SHCI_C2_Config(&config_param);

  APP_THREAD_Init();
  UTIL_LPM_SetStopMode(1 << CFG_LPM_APP, UTIL_LPM_ENABLE);
  UTIL_LPM_SetOffMode(1U << CFG_LPM_APP, UTIL_LPM_ENABLE);
  return;
}
```

RJMSTM, 2024-02-02:
ST Internal Reference: 172491

RJMSTM, 2024-02-20:
Hello @agalliazzo,

Excuse this delayed reply. Our development teams are already having a look to what you reported SRP is normally working fine but there are two important comments when looking at the code:

- NVM is saved before doing all the SRP stuff, so the SRP key cannot be stored. Please save NVM AFTER SRP has been registered successfully, you should be able to get the key back when restoring data from NVM.

- OT_NvmRamBuffer should absolutely NOT be placed in MB_MEM2, I don't know where you find the information that tell to do so but please tell me, so we can remove this wrong information. You mentionned Thread_SED_Coap_FreeRTOS\Core\Inc\hw_c but it doesn't exist.

Replace this:
`PLACE_IN_SECTION("MB_MEM2") uint32_t OT_NvmRamBuffer[CFG_THREAD_NVM_RAM_SIZE_IN_BYTES / 4];`

by this:
`uint32_t OT_NvmRamBuffer[CFG_THREAD_NVM_RAM_SIZE_IN_BYTES / 4];`

If you save just data in NVM just after being connected to the leader which is SRP server, but before SRP registration, it will fail on reboot and I get error 28 (OT_ERROR_RESPONSE_TIMEOUT). This is due to the key not being the good one, as it wasn't saved in NVM.

If you save after SRP registration, it works fine every time. you can sometimes get the error 29 (OT_ERROR_DUPLICATED) as SRP registration already exist on server's side but it immediately gives a retry knowing that and is successful.

which error you gets?

With Regards,
Rania

RJMSTM, 2024-03-13:
Hello,

Any feedback please!!

closed 2024-04-03 by RJMSTM