STMicroelectronics/STM32CubeWB #87: User code in  BEGIN SVCCTL_InitService */

TheAlbertDev, 2024-02-14:
**Describe the set-up**
 * STM32WB5MMG-DK
 * STM32CubeIDE 1.14.1
 * STM32Cube_FW_WB 1.17.3

**Describe the bug**
User code between `/* USER CODE BEGIN SVCCTL_InitService */` and `/* USER CODE END SVCCTL_InitService */` is erased after each code generation.

I want to increase manually the `max_attr_record` of a service to create manually a characteristic (the OTA characteristic following the heart rate OTA example). Code example:

```c
  /* USER CODE BEGIN SVCCTL_InitService */
  /* max_attr_record to be updated if descriptors have been added */
#if (BLE_CFG_OTA_REBOOT_CHAR != 0)
  max_attr_record += 2U;
#endif
  /* USER CODE END SVCCTL_InitService */
```

**How To Reproduce**
1. Configure in STMCubeMX the server app with custom template.

2. Create one service with one characteristic.

3. In code, add an additional characteristic to the service being the OTA characteristic (like in the heart rate OTA example provided by the package).

**Additional context**
N/A

**Screenshots**
N/A

RJMSTM, 2024-02-14:
Hello @TheAlbertDev,

Thank you for this report, can you try with STM32CubeWB v1.18.0 version. The error should disappear once you will upgrade to this version. 

Thank you again for your contribution.
With regards,

RJMSTM, 2024-03-14:
Hello
Any feedback please

closed 2024-04-04 by RJMSTM

TheAlbertDev, 2024-04-18:
Hello,

I have tried to upgrade to the new version, but it does not prompt me to upgrade (the typical continue or migrate message). In the .ioc it is not set to not ask .

```ProjectManager.AskForMigrate=true```

And in the project manager I don't have the option to change package (it doesn't even look like the typical default user check). 

[image]

How can I update the project?