STMicroelectronics/STM32CubeWB #89: CFG_LPM_SUPPORTED Abnormal during connection, There is a probability of getting stuck when exiting stop2 mode.

IC-killer, 2024-04-08:
Describe the set-up
STM32WB55CGU6 (Custom board)
STM32Cube_FW_WB_V1.17.3 (stm32wb5x_BLE_Stack_full_fw.bin, stm32wb5x_FUS_fw.bin)

**Describe the bug**
when I enable CFG_LPM_SUPPORTED,
there is a probability that the Bluetooth connection will get stuck at LL_HSEM_ReleaseLock(HSEM, CFG_HW_ENTRY_STOP_MODE_SEMID, 0) when exit stop2 mode.

I found that this link has the same issue as mine. [link](https://community.st.com/t5/stm32-mcus-wireless/can-t-lock-the-rcc-semaphore-when-exiting-low-power-mode-with/td-p/133526), 
but my flash and CPU2 clock frequencies are the same,,
[image]

this issue does not occur when Bluetooth is not connected.
when Bluetooth is connected, the program randomly gets stuck here and cannot recover normally.
stm32_lpm_if.c
```c
while(LL_HSEM_1StepLock(HSEM, CFG_HW_RCC_SEMID));
```
I have already disabled user tasks that are not related to Bluetooth.

ALABSTM, 2024-04-12:
Hi @IC-killer,

Thank you for these details. Two question though before digging deeper into this point:
* You mention you are using version **1.17.3** of the STM32CubeWB firmware. Current version is **1.19.0**. Could you please give it a try and see whether there is any issue yet?
* You use the expression "There is a probability". Have you experienced any actual issue or is it a guess based on code review, for instance?

Thank you in advance for the clarifications.

With regards,

IC-killer, 2024-04-13:
I have also tested versions 1.18 and 1.19, and encountered the same issue. 
This problem will definitely occur after a period of operation, but the length of time is uncertain. I don't know what possible reasons might have led to this. Modifying the UTIL_SEQ_Idle() function as in the link below can improve the situation: [[www.xxx.com](https://community.st.com/t5/stm32-mcus-wireless/ble-low-power-mode-stop2/td-p/240754)](https://community.st.com/t5/stm32-mcus-wireless/ble-low-power-mode-stop2/td-p/240754)
Thank you for your reply.

TOUNSTM, 2024-07-29:
Hello @IC-killer,

Could you please specify the IDE and its version, along with the smartphone application you are using?

With regards,

IC-killer, 2024-08-07:
@TOUNSTM MDK v5.37, cubeMX v6.9.2 and app use ST BLE ToolBox

TOUNSTM, 2024-08-29:
Hello @IC-killer,

Our development team would appreciate more information about your project. Specifically, it would be helpful to understand your use case in greater detail. For example, it is important to know if you are performing writes or erases in flash memory. You might find it useful to compare your code with the **BLE_HeartRate** application provided in the **STM32CubeWB** package, which uses **STOP2** mode.

Kind regards,

TOUNSTM, 2024-10-14:
Hello @IC-killer,

Thank you once more for your contribution. However, without additional details, we won't be able to investigate this issue further. Therefore, I will close this thread due to inactivity. You are welcome to reopen it at any time if you have more information to provide that can assist us in resolving this issue.
Thank you for your understanding.

With regards,

closed 2024-10-14 by TOUNSTM