STMicroelectronics/STM32CubeWB #90: Incompatible wide enumeration types

IC-killer, 2024-04-10:
**Describe the set-up**
STM32WB55CGU6 (Custom board)
STM32Cube_FW_WB_V1.17.3 (stm32wb5x_BLE_Stack_full_fw.bin, stm32wb5x_FUS_fw.bin)
keil MDK v5.37

**Describe the bug**
Incompatible wide enumeration types
The initialization of Bluetooth using wchar will not be successful.

I found the exception here:
Application/User/Core/app_entry.c --> static void APPE_SysEvtReadyProcessing(void * pPayload),
p_sys_ready_event->sysevt_ready_rsp is an enum type, I have already processed it with &0xff, but I'm not sure if there are any other impacts.

**How To Reproduce**
Uncheck this:
[image]

**Additional context**
```c
static void APPE_SysEvtReadyProcessing(void * pPayload)
{
  TL_AsynchEvt_t *p_sys_event;
  SHCI_C2_Ready_Evt_t *p_sys_ready_event;

  SHCI_C2_CONFIG_Cmd_Param_t config_param = {0};
  uint32_t RevisionID=0;
  uint32_t DeviceID=0;

  p_sys_event = (TL_AsynchEvt_t*)(((tSHCI_UserEvtRxParam*)pPayload)->pckt->evtserial.evt.payload);
  p_sys_ready_event = (SHCI_C2_Ready_Evt_t*) p_sys_event->payload;

  if ((p_sys_ready_event->sysevt_ready_rsp&0xFF) == WIRELESS_FW_RUNNING)
  {
    /**
    * The wireless firmware is running on the CPU2
    */
    APP_DBG_MSG(">>== WIRELESS_FW_RUNNING \n");

    /* Traces channel initialization */
    APPD_EnableCPU2();

    /* Enable all events Notification */
    config_param.PayloadCmdSize = SHCI_C2_CONFIG_PAYLOAD_CMD_SIZE;
    config_param.EvtMask1 = SHCI_C2_CONFIG_EVTMASK1_BIT0_ERROR_NOTIF_ENABLE
      +  SHCI_C2_CONFIG_EVTMASK1_BIT1_BLE_NVM_RAM_UPDATE_ENABLE
        +  SHCI_C2_CONFIG_EVTMASK1_BIT2_THREAD_NVM_RAM_UPDATE_ENABLE
          +  SHCI_C2_CONFIG_EVTMASK1_BIT3_NVM_START_WRITE_ENABLE
            +  SHCI_C2_CONFIG_EVTMASK1_BIT4_NVM_END_WRITE_ENABLE
              +  SHCI_C2_CONFIG_EVTMASK1_BIT5_NVM_START_ERASE_ENABLE
                +  SHCI_C2_CONFIG_EVTMASK1_BIT6_NVM_END_ERASE_ENABLE;

    /* Read revision identifier */
    /**
    * @brief  Return the device revision identifier
    * @note   This field indicates the revision of the device.
    * @rmtoll DBGMCU_IDCODE REV_ID        LL_DBGMCU_GetRevisionID
    * @retval Values between Min_Data=0x00 and Max_Data=0xFFFF
    */
    RevisionID = LL_DBGMCU_GetRevisionID();

    APP_DBG_MSG(">>== DBGMCU_GetRevisionID= %lx \n\r", RevisionID);

    config_param.RevisionID = (uint16_t)RevisionID;

    DeviceID = LL_DBGMCU_GetDeviceID();
    APP_DBG_MSG(">>== DBGMCU_GetDeviceID= %lx \n\r", DeviceID);
    config_param.DeviceID = (uint16_t)DeviceID;
    (void)SHCI_C2_Config(&config_param);

    APP_BLE_Init();
    UTIL_LPM_SetOffMode(1U << CFG_LPM_APP, UTIL_LPM_ENABLE);
  }
  else if ((p_sys_ready_event->sysevt_ready_rsp&0xFF) == FUS_FW_RUNNING)
  {
    /**
    * The FUS firmware is running on the CPU2
    * In the scope of this application, there should be no case when we get here
    */
    APP_DBG_MSG(">>== SHCI_SUB_EVT_CODE_READY - FUS_FW_RUNNING \n\r");

    /* The packet shall not be released as this is not supported by the FUS */
    ((tSHCI_UserEvtRxParam*)pPayload)->status = SHCI_TL_UserEventFlow_Disable;
  }
  else
  {
    APP_DBG_MSG(">>== SHCI_SUB_EVT_CODE_READY - UNEXPECTED CASE \n\r");
  }

  return;
}
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

ALABSTM, 2024-04-12:
Hi @IC-killer,

As you mentioned, structure member `sysevt_ready_rsp` is an enumeration, as shown below.

https://github.com/STMicroelectronics/STM32CubeWB/blob/1ceb3b0c5bfffbf0ee20bccec683f37a0c2a51d5/Middlewares/ST/STM32_WPAN/interface/patterns/ble_thread/shci/shci.h#L34-L38

Its value is checked against the two values the enumerated type could take, as shown below.

https://github.com/STMicroelectronics/STM32CubeWB/blob/1ceb3b0c5bfffbf0ee20bccec683f37a0c2a51d5/Projects/NUCLEO-WB15CC/Applications/BLE/BLE_HeartRate/Core/Src/app_entry.c#L476

https://github.com/STMicroelectronics/STM32CubeWB/blob/1ceb3b0c5bfffbf0ee20bccec683f37a0c2a51d5/Projects/NUCLEO-WB15CC/Applications/BLE/BLE_HeartRate/Core/Src/app_entry.c#L517

You mentioned an **exception**. Could you please share the error message you got please? Thank you.

By the way, you mention you are using version **1.17.3** of the STM32CubeWB firmware. Current version is **1.19.0**.

With regards,

IC-killer, 2024-04-13:
After using an enumeration type with the int width, the program fails to enter the correct if statement branch. 

When the execution reaches this point, the value of `p_sys_ready_event->sysevt_ready_rsp` is 0x38000000.  I'm not sure about the meaning of the high byte 0x38, it exceeds the range of a char-width enumeration type. so I temporarily dealt with it using "&0xFF". Version 1.19 has the same issue.

Thank you for your reply.

closed 2024-05-08 by ALABSTM