STMicroelectronics/STM32CubeWB #91: Documentation issues

marc6h, 2024-05-17:
There is some unclear/sloppy documentation in this repo, [such as here](https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Projects/NUCLEO-WB15CC/Applications/FreeRTOS/FreeRTOS_Mutexes/Src/main.c). What is meant by "this tadhr" or "this Take"? "Through the mutex" means what?

```
    /* The first time through the mutex will be immediately available, on
    subsequent times through the mutex will be held by the low priority thread
    at this point and this Take will cause the low priority thread to inherit
    the priority of this tadhr.  In this case the block time must be
    long enough to ensure the low priority thread will execute again before the
    block time expires.  If the block time does expire then the error
    flag will be set here */
    if (osMutexWait(osMutexHandle, mutexTWO_TICK_DELAY) != osOK)
    {
      /* Toggle LED3 to indicate error */
      BSP_LED_Toggle(LED3);
    }
```

ALABSTM, 2024-05-21:
Hi @marc6h,

Thank you for this report. Please note the word "through" is part of both expressions "the first time through" and "on subsequent times through".

Regarding the word "tadhr", it is likely to be a typo. I will ask to check the relevance of the word "Take". I will keep you informed.

With regards,

ALABSTM, 2024-05-27:
See also [STM32CubeG4#44](https://github.com/STMicroelectronics/STM32CubeG4/pull/44)

ALABSTM, 2024-07-12:
Fixed in commit fac4f05637d6444c1f624ec4ec70d785466789fc

closed 2024-07-12 by ALABSTM