STMicroelectronics/STM32CubeWB #93: STM32WB aci_gap_create_connection return 0x86

huangliguang, 2024-06-17:
我是中国人

use 1 master for 8 slave

1 slave disconneted ，and master connect the disconnect slave ，but aci_gap_create_connection return 0x86 ，
Is it because resources are not released?

RJMSTM, 2024-06-20:
ST Internal Reference: 184655

AppliBLE, 2024-06-25:
Hi, there,
the error code 0x86 corresponds to "Slot length failed". 

Can you give us more info about your project / use case?
Which project did you start from?
What is the connection interval value?
Does this happen every time?
When you disconnect the slave, do you receive the complete disconnection event?

BR, Joé

huangliguang, 2024-06-26:
mater conn param
result = aci_gap_create_connection(SCAN_P,
                                               SCAN_L,
                                               server_Remote_Addr[i].adv_t.SERVER_REMOTE_ADDR_TYPE, server_Remote_Addr[i].adv_t.SERVER_REMOTE_BDADDR,
                                               CFG_BLE_ADDRESS_TYPE,
                                               CONN_P(68),
                                               CONN_P(68),
                                               0,
                                               SUPERV_TIMEOUT,
                                               CONN_L(6),
                                               CONN_L(6));

This connection is just enough to connect eight slaves, not one more. But connected 8 slave machine, there is a slave machine I directly shut down, again connected to the broken line of the slave machine can not be connected, reported 0x86

 

客户
***@***.***

RJMSTM, 2024-07-09:
Please any feedback!!

huangliguang, 2024-07-15:
[v1.19.1](https://github.com/STMicroelectronics/STM32CubeWB/releases/tag/v1.19.1)  this version

params
                result = aci_gap_create_connection(SCAN_P,
                                                   SCAN_L,
                                                   server_Remote_Addr[i].adv_t.SERVER_REMOTE_ADDR_TYPE, server_Remote_Addr[i].adv_t.SERVER_REMOTE_BDADDR,
                                                   CFG_BLE_ADDRESS_TYPE,
                                                   CONN_P(70),
                                                   CONN_P(70),
                                                   0,
                                                   SUPERV_TIMEOUT,
                                                   CONN_L(6),
                                                   CONN_L(6));

AppliBLE, 2024-08-06:
Hi @huangliguang , 
In the previous comments, I have the impression that you have different values for the aci_gap_create_connection, connection interval 68 / 70.
If you make all your connections with a certain CI, when you disconnect and reconnect to your slave you must use the same CI or a multiple of it.
The error could come from there.

If you use the same interval connection all the time, the it is another problem.
We'll need more information about your project to debug it.

BR, Joé

huangliguang, 2024-08-09:
> Hi @huangliguang , In the previous comments, I have the impression that you have different values for the aci_gap_create_connection, connection interval 68 / 70. If you make all your connections [...]

all then same conn_param
`result = aci_gap_create_connection(SCAN_P,
								   SCAN_L,
								   server_Remote_Addr[i].adv_t.SERVER_REMOTE_ADDR_TYPE, server_Remote_Addr[i].adv_t.SERVER_REMOTE_BDADDR,
								   CFG_BLE_ADDRESS_TYPE,
								   CONN_P(82),
								   CONN_P(82),
								   0,
								   SUPERV_TIMEOUT,
								   CONN_L(8),
								   CONN_L(8));`

1 mater and 8 slave
and 1 slave disconnect,then msater start scan discovery,but retrun 0x86

AppliBLE, 2024-08-19:
Hi @huangliguang 
We can't reproduce the problem, 
would it be possible for you to share your project so that we can debug?

BR, Joé

huangliguang, 2024-09-29:
A host connects to eight slaves, after all are connected, randomly disconnects one slave, and then opens the scan, but returns 0x86

huangliguang, 2024-09-29:
```               
 result = aci_gap_create_connection(SCAN_P,
                                                   SCAN_L,
                                                   SERVER_REMOTE_ADDR_TYPE, 
                                                   SERVER_REMOTE_BDADDR,
                                                   CFG_BLE_ADDRESS_TYPE,
                                                   CONN_P(113),
                                                   CONN_P(113),
                                                   0,
                                                   SUPERV_TIMEOUT,
                                                   CONN_L(12),
                                                   CONN_L(12));
```
connect param