STMicroelectronics/STM32CubeWB #94: Hardfault when using the sequencer with a number of priorities higher than 2

TheAlbertDev, 2024-06-18:
**Describe the set-up**
 * STM32WB5MM-based custom board.
 * STM32CubeIDE Version: 1.15.1 Build: 21094_20240412_1041 (UTC)
 * STM32Cube FW_WB V1.19.0

**Describe the bug**
When additional priorities are configured in the sequencer, the sequencer throws a hardfault.
```c 
/**
 * This is the list of priority required by the application
 * Each Id shall be in the range 0..31
 */
typedef enum
{
  CFG_SCH_PRIO_0,
  /* USER CODE BEGIN CFG_SCH_Prio_Id_t */
  CFG_SCH_PRIO_1,
  CFG_SCH_PRIO_2,
  CFG_SCH_PRIO_3,
  CFG_SCH_PRIO_4,
  /* USER CODE END CFG_SCH_Prio_Id_t */
} CFG_SCH_Prio_Id_t;
```

You can edit manually `UTIL_SEQ_CONF_PRIO_NBR` to the new number of priorities, but as soon as the project code is regenerated, `UTIL_SEQ_CONF_PRIO_NBR` is set to 2 again.

As stated in issues in another packages (https://github.com/STMicroelectronics/STM32CubeWL/issues/18), a solution could be to define `UTIL_SEQ_CONF_PRIO_NBR` as `CFG_SEQ_Prio_NBR`, and in the priorities section define:
```diff
{
  CFG_SCH_PRIO_0,
  /* USER CODE BEGIN CFG_SCH_Prio_Id_t */
  CFG_SCH_PRIO_1,
  CFG_SCH_PRIO_2,
  CFG_SCH_PRIO_3,
  CFG_SCH_PRIO_4,
  /* USER CODE END CFG_SCH_Prio_Id_t */
+ CFG_SEQ_Prio_NBR
} CFG_SCH_Prio_Id_t;
```

RJMSTM, 2024-06-20:
ST Internal Reference: 184654

RJMSTM, 2024-12-02:
Hello,
I hope you are fine. The issue you reported has been fixed in the frame of version v1.21.0 of the STM32CubeWB published recently on GitHub. Thank you again for having reported.

closed 2024-12-02 by RJMSTM