STMicroelectronics/STM32CubeWB #95: ble_gatt_aci.h missing C++ support

raitraak-rrk, 2024-07-14:
Middlewares/ST/STM32_WPAN/ble/core/auto/ble_hal_aci.h is missing C++ support. Many other STM32_WPAN files look the same.

#ifdef __cplusplus
extern "C" {
#endif

RJMSTM, 2024-07-16:
ST Internal Reference: 186693

RJMSTM, 2024-08-07:
Hello,

Thank you for your contribution. You need to include "Middlewares\ST\STM32_WPAN\ble\ble.h" before BLE_ACI.h.

Regards,

raitraak-rrk, 2024-08-08:
Thanks. However, this seems more of a undocumented workaround rather than a fix.

RJMSTM, 2024-10-02:
Hello,

A cleaner solution is to select the STM32_WPAN group in Middlewares and go in "Options" panel, "C/C++ Compiler" and select "Override inherited settings" and select C language.

Regards,

raitraak-rrk, 2024-11-01:
There is no "Options" panel when using git repo directly.

closed 2024-12-02 by RJMSTM