STMicroelectronics/STM32CubeWB #96: aci_gap_start_connection_update return 0x12

huangliguang, 2024-08-09:
ble version：v1.20.0
`                result = aci_gap_create_connection(SCAN_P,
                                                   SCAN_L,
                                                   server_Remote_Addr[i].adv_t.SERVER_REMOTE_ADDR_TYPE, server_Remote_Addr[i].adv_t.SERVER_REMOTE_BDADDR,
                                                   CFG_BLE_ADDRESS_TYPE,
                                                   CONN_P(82),
                                                   CONN_P(82),
                                                   0,
                                                   SUPERV_TIMEOUT,
                                                   CONN_L(8),
                                                   CONN_L(8));`

conn success then call aci_gap_start_connection_update 
`result = aci_gap_start_connection_update(server_Remote_Addr[i].BleApplicationContext_legacy.connectionHandle,
                         CONN_P(82),
                         CONN_P(82),
                         10,    //Connection_Latency = ((connSupervisionTimeout / (connInterval*2)) - 1)。
                         150,  // Supervision_Timeout = (1 + connSlaveLatency) * connInterval * 2。
                         CONN_L(8), CONN_L(8));`

but retrun 0x12

test is  1 master and 1 slave

ALABSTM, 2024-08-09:
Hi @huangliguang,

Would you mind providing more details about this issue. A template is provided to guide you. This would help us supporting you.

With regards,

ALABSTM, 2024-09-05:
Hi @huangliguang,

Please allow me to close this thread as no feedback so far. Thank you for your comprehension.

With regards,

closed 2024-09-05 by ALABSTM