STMicroelectronics/STM32CubeWB #101: aci_gap_start_general_discovery_proc return 0x86

huangliguang, 2024-08-26:
1、1master use the same conn param to connect 8 slave
       Conn_Interval_Min = CONN_P(113)
       Conn_Interval_Max= CONN_P(113)
      Minimum_CE_Length =CONN_L(12)
      Maximum_CE_Length = CONN_L(12)
2、call aci_gap_terminate Randomly disconnecting four of them
3、The connection parameters of the remaining four slave were updated
aci_gap_start_connection_update(server_Remote_Addr[i].BleApplicationContext_legacy.connectionHandle,
                         CONN_P(57),
                         CONN_P(57),
                         0,
                         SUPERV_TIMEOUT,
                         CONN_L(12),
                         CONN_L(12);
4、after update success，Update parameters again
aci_gap_start_connection_update(server_Remote_Addr[i].BleApplicationContext_legacy.connectionHandle,
                         CONN_P(113),
                         CONN_P(113),
                         0,
                         SUPERV_TIMEOUT,
                         CONN_L(12),
                         CONN_L(12);
5、Enable scan ，then return 0x86
aci_gap_start_general_discovery_proc

RJMSTM, 2024-08-30:
Hello @huangliguang,

Would you mind providing more details about this issue. A template is provided to guide you. This would help us supporting you.

With regards,

huangliguang, 2024-08-30:
> Hello @huangliguang,
> 
> Would you mind providing more details about this issue. A template is provided to guide you. This would help us supporting you.
> 
> With regards,

My function is too closely related to the project, if there is no related peripherals, it will run, so you need to build your own project test according to the above parameters, this problem will occur 100%

huangliguang, 2024-10-28:
Ble version ： 1.20
1、1master use the same conn param to connect 8 slave
Conn_Interval_Min = CONN_P(113)
Conn_Interval_Max= CONN_P(113)
Minimum_CE_Length =CONN_L(12)
Maximum_CE_Length = CONN_L(12)
2、call aci_gap_terminate disconnect the first connect Slave
3、Enable scan ，then return 0x86 （slot length fail）
aci_gap_start_general_discovery_proc((0x320), (0x320), CFG_BLE_ADDRESS_TYPE, 1);

huangliguang, 2024-11-18:
@RJMSTM @ALABSTM @TOUNSTM @avisconti

huangliguang, 2024-11-18:
The middle slot is disconnected and will return 0x86 if you want to scan, but the last 8 connection is disconnected and will not return 0x86 when you start scanning

huangliguang, 2025-01-17:
@RJMSTM @ALABSTM @TOUNSTM @avisconti Can you fix this bug? Our product is using this chip, once disconnected, it will never be connected, you must power off and power back on

RJMSTM, 2025-01-30:
Hello @huangliguang,

First, I apologize for the delayed response.
Second, refer to the Programming Manual PM0271 [Guidelines for Bluetooth® Low Energy stack programming on STM32WB/
STM32WBA MCUs](https://www.st.com/resource/en/programming_manual/pm0271-guidelines-for-bluetooth-low-energy-stack-programming-on-stm32wb-stm32wba-mcus-stmicroelectronics.pdf)

**Section 6.3.1**.: As explained in the documentation, the scan Interval should be equal to the connection interval of one of the existing central connections.

[image]

In your case the Scan Interval (0x320) is greater than the connection interval, that's why the command return 0x86. 

I hope this helps you.

Regards,
Rania

RJMSTM, 2025-04-25:
Hello,

Thank you once again for your contribution. I hope the response was helpful to you. Please allow me to close the ticket. Otherwise, we are still awaiting your contribution